ArduPilot/ardupilot #19362: waf: `sizeof-pointer-div` causes build failures on Ubuntu 18.04

sypaq-nexton, 2021-11-25:
## Bug report

**Issue details**

PR #19351 added the `-Werror=sizeof-pointer-div` compiler option when building for `gcc` versions greater than or equal to v7.4. However, this option is not available in v7.4. As such, a build on the Ubuntu 18.04 LTS (which by default uses `gcc` v7.5) will fail.

Note: I haven't confirmed _exactly_ which `gcc` includes the `sizeof-pointer-div` option:
* it _is not_ present in v7.5
* it _is_ present in v8.4

**Version**
`master`

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

sypaq-nexton, 2021-11-25:
I'll raise a PR shortly...

peterbarker, 2021-11-26:
PR merged - thanks!

closed 2021-11-26 by peterbarker