ArduPilot/ardupilot #19368: Rover: compensate for asymetric thrust from propellers.

geofrancis, 2021-11-25:
When testing Arduboat I ran into a problem where the controller doesn't take into account the difference in thrust when running a propeller forwards vs reverse. The result is when trying to yaw the boat with a left and right motor the boat will shoot forwards and performing a wide circle rather than turning on the spot since the inboard motor that's running in reverse just can't overcome the thrust of the forward running propeller, this is more noticeable the narrower the boat since there is less leverage, the way I compensated for it was to just limit the forward thrust using the pwm limits but with Dshot that doesn't work any more, and it still restricts forward thrust. 

I tried testing with the current limiters set on the ESCs but they are using the same amount of power and the difference is in the efficiency of the propellers, so it was still going forwards. 

What is needed is a percentage setting for how much of maximum forward thrust is used when turning with the left and right motors.

rmackay9, 2021-11-30:
I think this is a duplicate of https://github.com/ArduPilot/ardupilot/issues/8800

geofrancis, 2021-11-30:
> I think this is a duplicate of #8800

Its related, #8800 is describing controlling a vehicle with a asymmetric throttle/reverse speed range. with extra limits for reverse,  augmenting the speed control of a vehicle that can go backwards and forwards, its not for augmenting steering control on a skid steer boat.  for this issue we only need the limits when steering, general back and forward speed control is not the issue.

geofrancis, 2022-07-07:
I just found a case where this is a real problem. When testing bendy ruler the vehicle wants to hold position and rotate so it can scan for a path, that won't be possible if it can't stay still and turn at the same time because it will shoot forward if it tries.

geofrancis, 2022-09-25:
I have seen reports from in the wild now of people having real problems trying to rotate on the spot and have had to 3d print symmetrical propellers to compensate.

rmackay9, 2023-09-06:
This is include in 4.4.0-beta6 so I will close this now, thanks!

closed 2023-09-06 by rmackay9