ArduPilot/ardupilot #19376: IMU_FAST ArduPilot Gyroflow

TMac54, 2021-11-27:
## Feature request or bug?

Copter 4.1.1 Stable:  No IMU log message with IMU_FAST enabled (for Gyroflow use)
After enabling IMU_FAST within LOG_BITMASK for logging IMU data, no IMU dataflash log message is captured.

Log file with no IMU data captured after IMU_FAST enabled is [here](https://bit.ly/LogforNoIMUMsgwith_IMU_FAST)
Matek H743 Mini flight controller and SanDisk 32Gb A1 micro SD card formatted with Fat32 (works fine with logging other messages).

ATT_FAST as an alternative is not supported by Gyroflow video stabilization software which requires IMU msg data/format.

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Have toggled on IMU_FAST option for LOG_BITMASK within Mission Planner 1.3.75.1 build 1.3.7990.37522 both with and without (value = 262144) other LOG_BITMASK options enabled.  Same result = no IMU message captured when attempting to display with Mission Planner and also with UAV Log Viewer.

I have submitted this on the Copter 4.1 Support Forum (on 26 November) with no replies to date prior to opening this issue is [here](https://bit.ly/SupportForumPostIMU_FAST)

Attempting to use IMU_FAST as a "faster log rate" for the gyro data than the normal IMU log rate (although I am unaware of the specific log rate for IMU_FAST).  Purpose is for use with Gyroflow video stabilization software developed by Elvin Chen.  This was previously discussed on ArduPilot Discord #General with Peter Hall is [here](https://bit.ly/LogRateDiscordDiscussion)
Gyroflow Discord site is RSGoBlackbox > Gyroflow Dev is [here](https://discord.com/channels/649358865070161920/762689285353635900)

Elvin Chen has done preliminary work for Gyroflow to support ArduPilot IMU log messages.  Gyroflow ideally requires IMU data in at least 100 hz rate (400 hz would be great) and in format of IMU message.  Then, with Mission Planner, Data Tab, CNTRL F, DashWare, select log file, Enter "IMU" only as message you want.  This then creates a  .bin.CSV file format with IMU data which can be used by Gyroflow.  

IMU message format with 400 hz log rate would enable Gyroflow video stabilization software support of ArduPilot firmware logging.  This would give users of ArduPilot the ability to stabilize flight videos using Gyroflow free video stabilization software with just about any camera onboard simply by using the gyro data from the IMU message with a fast logging rate.

peterbarker, 2021-12-01:
We appear to have broken proper fast-IMU-logging in 2016....

```
commit b2353042359c5fc2e44bb05d651fe1321ab822a1
Author: Andrew Tridgell <andrew@tridgell.net>
Date:   Tue May 3 13:21:23 2016 +1000

    Copter: reduce IMU logging to 25Hz
    
    leave more room for IMT logging needed for Replay
```

peterbarker, 2021-12-01:
@TMac54 is there some reason you want to use the IMU message for this?  These are raw sensor readings - data is present for more than one IMU sensor, and you won't be able to easily tell which if the IMUs is actually worth listening to trivially.

AFAICS you should use the `RATE` message (https://ardupilot.org/copter/docs/logmessages.html#rate-desired-and-achieved-vehicle-attitude-rates-not-logged-in-fixed-wing-plane-modes) - it can be output at 400Hz using the `ATTITUDE_FAST` bit.

TMac54, 2021-12-01:
> @TMac54 is there some reason you want to use the IMU message for this? These are raw sensor readings - data is present for more than one IMU sensor, and you won't be able to easily tell which if the [...]

Thanks for your response Peter.  I'm not the Gyroflow Dev, so I'm not the best one to answer your question.  Elvin Chen is the main Dev of gyroflow.  I believe it's due to the format.  The RATE  message gives "gyro info" on the Roll, Pitch, and Yaw axes in deg/sec whereas the IMU message gives Gyr X, Gyr Y, and Gyr Z info in rad/sec with the Gyroflow reference axes as shown in image linked [here](https://drive.google.com/file/d/1p-vLfWuGYEmLwzJeiyJvmLI8CUYfsBiX/view?usp=sharing).  Gyroflow uses gyro data to stabilize video footage.  He has implemented preliminary support of ArduPilot within it by using the "regular" IMU message (using it's format).  If there's a way to implement the filtered IMU message format at a logging rate of greater than 100 hz (preferably 400-500 hz?), that would be ideal.    Elvin  (and other Gyroflow Dev's) can be reached at the Gyroflow Dev Discord site linked [here](https://discord.com/channels/649358865070161920/762689285353635900)

IamPete1, 2021-12-03:
@ElvinC  Does the rate message have what you need? https://ardupilot.org/copter/docs/logmessages.html#rate-desired-and-achieved-vehicle-attitude-rates-not-logged-in-fixed-wing-plane-modes You can get it at full loop rate (400hz by default) by setting the fast attitude logging bit.

If you can tell us what you need we could add a new GyroFlow logging message. There are lots of extra things in fast attitude logging you would not need and result in massive log files.

ElvinC, 2021-12-03:
Oh hey, thanks for the ping. TMac sent a file with the rate messages yesterday that I'll have a look at, but based on the docs it should work. For gyroflow use, the following should be everything:
* us timestamp
* Filtered angular velocities around local axes in x,y,z, which the RATE message includes
* Measured acceleration around local axes (if available)
* Measured magnetic field vector around local axes (if available)

ElvinC, 2021-12-03:
I can now confirm that the roll/pitch/yaw rate data in the RATE messages work as expected with the files sent by @TMac54

To elaborate on the previous message, the minimum required are timestamps and angular rates. Adding accelerometer allows for horizon leveling as you'd expect. The magnetometer currently doesn't have a significant influence (drift doesn't matter for stabilization), so excluding this one would make sense if longer logs are desired.

https://user-images.githubusercontent.com/20195216/144685321-cbd84f87-1681-43a6-a880-295fa60a7f56.MP4

TMac54, 2021-12-04:
Here's a short "How to", [2-minute video clip](https://bit.ly/HowTo_Rate_dot_CSV) on the process used to get the .CSV RATE log from the original .bin log file.
The .CSV RATE file is what was used in the Dev version of Elvin's Gyroflow for the 4 sec vid clip posted by ElvinC and currently supported by Dev version of Gyroflow.  See attached image. Many thanks to @peterbarker , @IamPete1 , and @ElvinC !  Believe this to be a "Win-Win" for both ArduPilot & Gyroflow!
[image]

IamPete1, 2022-09-04:
Closed by https://github.com/ArduPilot/ardupilot/pull/19531, I think Gryoflow now supports AP `.bin` files directly.

closed 2022-09-04 by IamPete1