ArduPilot/ardupilot #19382: Nan and other failure cases handelling

shubham-shahh, 2021-11-28:
## Bug report

Hi, I have described the issue[ here](https://github.com/mavlink/mavros/issues/1647)

_Please describe the problem_

The copter tends to crash when a path planner passes "Nan" values. 
**Version**
_What version was the issue encountered with_
4.0.5
**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Pixhawk (SITL)

**Logs**
NA

magicrub, 2021-11-28:
Can you be more specific?

shubham-shahh, 2021-11-28:
Hi,
the external path planners sometimes give "nan" values for yaw-rate (for example fast planner) if we convert that message to mavros type and pass it to ardupilot the copter firmware crashes (tested it in SITL).
I don't have logs but if logs are required I'll reproduce the issue once again

thanks

rmackay9, 2021-11-30:
@shubham-shahh which mavlink messages in particular are you referring to?  Here are some choices:

- SET_POSITION_TARGET_LOCAL_NED
- SET_POSITION_TARGET_GLOBAL_INT
- SET_ATTITUDE_TARGET

shubham-shahh, 2021-11-30:
> @shubham-shahh which mavlink messages in particular are you referring to? Here are some choices:
> 
> * SET_POSITION_TARGET_LOCAL_NED
> * SET_POSITION_TARGET_GLOBAL_INT
> * SET_ATTITUDE_TARGET

Hi, @rmackay9, I am using /mavros/setpoint_raw/local of type mavros_msgs::PositionTarget which refers to the mavlink message SET_POSITION_TARGET_LOCAL_NED

shubham-shahh, 2021-11-30:
> @shubham-shahh which mavlink messages in particular are you referring to? Here are some choices:
> 
> * SET_POSITION_TARGET_LOCAL_NED
> * SET_POSITION_TARGET_GLOBAL_INT
> * SET_ATTITUDE_TARGET

Hi, whats your opinion? shall we add such failure cases in ardupilot? if yes Please let me know, I would love to contribute more @rmackay9

rmackay9, 2021-11-30:
@shubham-shahh, yes, I think if the field is used we should check for NaN and reject the message if it contains one (or more).

shubham-shahh, 2021-11-30:
> @shubham-shahh, yes, I think if the field is used we should check for NaN and reject the message if it contains one (or more).

Hi, thanks for the feedback. I was curious if we should do this with all MAVLINK messages?

I'll work on the correction and put up a PR.

Thanks

shubham-shahh, 2021-12-03:
> @shubham-shahh, yes, I think if the field is used we should check for NaN and reject the message if it contains one (or more).

I tired to re create the issue to solve it but looks like it is fixed

shubham-shahh, 2021-12-03:
> 

I was planning to add a fix [here](https://github.com/ArduPilot/ardupilot/blob/308f4e99b4d3cda98ad89529bfc3d3faaa800173/ArduCopter/GCS_Mavlink.cpp#L1217)