ArduPilot/ardupilot #19395: Rover: ground steering is inverted when reversing using PILOT_STEER_TYPE 3

geofrancis, 2021-11-29:
When reversing with thrusters and a rudder in PILOT_STEER_TYPE 3, I noticed the rudder was not pointing the right way, the flight controller is assuming that it's going to give  rotation the same direction regardless if its going forwards or backwards but a rudder doesnt work like that, the flow over it reverses and so does its force.

rmackay9, 2021-11-29:
@geofrancis,

Thanks for the report.

The PILOT_STEER_TYPE should only affect the transmitter stick input processing so it shouldn't depend upon what type of frame is being used... so if it's wrong it should be wrong for all frame types.

 Not saying there isn't a problem, it's just that it shouldn't be related to the PILOT_STEER_TYPE as far as I can imagine at least..

geofrancis, 2021-11-30:
I think the problem is from the mix of omni and regular controls, it's not something that you would run across on a wheeled rover and I dont think many used a Rudder with omni motors at the same time, I had to disable prearm checks just to get this configuration to work as it would fail on having multiple steering methods. PILOT_STEER_TYPE 0, the rudder acts as expected but the thrusters act strange as they will reverse as soon as you go from forwards to backwards to emulate a rudder but that is normal behaviour it just makes manoeuvring a little weird when your used to flying. With PILOT_STEER_TYPE 3 the thrusters work as expected almost like hovering a multirotor, but now the rudder acts against the thrusters when reversing since its effectively backwards because of the reverse flow over it. 

ground steering to the left with forward thrust will rotate the boat left but reversing main propulsion will rotate the boat to the right. Thrusters will rotate to the right or left regardless of direction of main propulsion, and its this conflict thats the issue.

rmackay9, 2021-11-30:
@geofrancis,

Ok, yes, I think you're right that the issue is the mixing of frames.  Rover doesn't support mixing ackerman and skid-steering (or omni) vehicles.  There is at least one other issue as well which is that the steering will not be scaled properly for one of the steering controls.  With an ackermann steering vehicle the response increases with vehicle speed but the compensation won't be applied if the vehicle is also a skid-steering vehicle.

I think we should rename this issue to be something like, "support mixed ackerman and skid-steering"

geofrancis, 2021-11-30:
ok that makes sense. I was also seeing some strange throttle and rudder movements when trying manoeuvres I thought that was a separate issue but its probably related to the incorrect scaling being applied.  Any thoughts on how to work around it ?

rmackay9, 2021-11-30:
@geofrancis,

I don't know of any work around I'm afraid.  It's an enhancement that's required to separate the motor types more completely within the motor library so that the scaling can be applied to each separately.

geofrancis, 2021-11-30:
any reason it cant run both?  I assume one is only ran  for performance reasons.

geofrancis, 2021-11-30:
I have never actually seen it in writing anywhere that you cant mix steering modes, is it in the wiki?

rmackay9, 2021-11-30:
@geofrancis,

No, I had a quick peek now as well and it does not mention that mixing steering configurations is not supported.  We could add a note .. you could argue that there's an infinite number of things that we don't support and listing them all would be impossible ... but perhaps mixing steering methods is something that numeous people might assume works when it doesn't actually.

geofrancis, 2021-12-01:
ok I understand. I have just never seen any mention of anyone ever having a problem so I just assumed it would just work, and it treated all the outputs the same, I didn't realize the mixing mode was tied to frame class, I thought frame class only selected what outputs it was using not the mixing and scaling too.

rmackay9, 2022-08-22:
@geofrancis, I've created a PR to remove the arming check for 4.3. https://github.com/ArduPilot/ardupilot/pull/21507

closed 2022-08-22 by geofrancis