ArduPilot/ardupilot #19417: NMEA 0183 data input function via serial port.

AlksSAV, 2021-12-02:
**Support for input from NMEA 0183 devices**

## Feature request

I would like the Ardupilot development team to add a new feature for the serial port.
This function of data input from NMEA 0183 devices would allow combining all devices via one Serial port using a multiplexer.
A similar parameter already exists in "SERIAL*_PROTOCOL", only there the data goes in the opposite direction.

**Platform**
[x] Rover

**Additional **
I have created a [discussion thread](https://discuss.ardupilot.org/t/support-for-input-from-nmea-0183-devices/79144), in which I will post those devices (with nmea support) that I could test. At the moment I have a compass, marine GPS, sonar.  nmea multiplexer will be available soon.
I 'm also doing a weather vane project .

rmackay9, 2022-07-22:
Txs for this.  I temporarily misunderstood the request so I changed the title twice.

AlksSAV, 2022-07-22:
> Txs for this.  I temporarily misunderstood the request so I changed the title twice.

The point is to receive data from all nmea devices via a single port.

madflower, 2022-08-14:
You will be better served using something like a Maxim MAX14830ETM+ which is a quad uart to spi or i2c chip which is about 10 bucks. A multiplexer, which might work, but it will be limited to the max bus speed of 115kbps IO of the master and the multiplexer and you may end up with dropped packets.  If you move to spi or i2c, then you increase the data transmission rate and you can have 4 devices that spew a lot more data like 6Mbps I suspect programmatically it would be a lot easier implement as well but I haven't looked at the code for it.

AlksSAV, 2022-08-15:
> You will be better served using something like a Maxim MAX14830ETM+ which is a quad uart to spi or i2c chip which is about 10 bucks. A multiplexer, which might work, but it will be limited to the [...]

The problem is not the speed of the autopilot port. The problem is that it cannot receive information in one port from all NMEA devices (gyrocompass, echo sounder, magnetic compass, AIS, weather vane, etc.)