ArduPilot/ardupilot #19424: Lua scripts cant get time in air

p8410077, 2021-12-03:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

Hi,
I have been trying to get "the time in air" and send the message to GCS via Lua.  
After I power up the aircraft, the message shows 

[image]

Lua: No scripts to run
Lua: No scripts to run
Lua: /APM/scripts/new 1.lua:2: attempt to call a n

The script was as below
 
[image]

**Version**
plane 4.0.9

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
qplane

**Hardware type**
cube blue

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2021-12-03:
@p8410077,  I think if you retry with Latest it might print a better error message.  I'm not totally sure but I think the script is failing for some reason but it is happening so early after startup that the reason is not displayed on the terminal.

IamPete1, 2021-12-03:
`get_time_flying_ms` is not available on 4.0, you will have to update to 4.1 to use it.

closed 2021-12-03 by IamPete1