ArduPilot/ardupilot #19427: AP_Proximity: Device checksum value is not wrong.

muramura, 2021-12-04:
## Bug report

**Issue details**

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Proximity/AP_Proximity_Cygbot_D1.cpp#L116
The size value is missing 1 byte in this process.

_Please describe the problem_

https://github.com/CygLiDAR-ROS/cyglidar_d1/blob/v0.1.3/src/CygbotParser.c#L57

The original calculates the checksum from Length1.
The ArduPilot process makes it calculate the checksum from Length1, Length2 and the payload only.
One byte of the payload header is missing.
The comment describes that the checksum does not match.
This is because of a mistake in the process.

**Version**
_What version was the issue encountered with_

ArduCopter V4.2.0-dev

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

quad

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2021-12-04:
This one is for @rishabsingh3003

rishabsingh3003, 2021-12-04:
Thanks, @muramura. I'll check this out and push a fix!

Ryanf55, 2023-03-14:
@muramura Does this solve your issue?

rishabsingh3003, 2023-03-14:
I believe I fixed this issue.

closed 2023-03-14 by rishabsingh3003