ArduPilot/ardupilot #19434: Support Common Smart BMS boards.

geofrancis, 2021-12-04:
I have been looking to find a BMS that is compatible with ardupilot and have been struggling to find anything that you can actually buy without emailing the company just to get a price. 

These BMS board are very popular in the Scooter and ebike community as they are reliable and totally programmable via bluetooth over uart. The protocol is documented so everything is there for it to be added. they are very reasonably priced and available from 3s to 30s

https://www.lithiumbatterypcb.com/product-category/smart-bms/

the protocol documentation can be found here 
https://www.lithiumbatterypcb.com/Protocol%20English%20Version.rar

hendjoshsr71, 2021-12-04:
@geofrancis Have you used any of these before? Are there other manufacturers that use this communication protocol?

Do you know of any repositories that have the protocol coded already?

I'm not a big fan of having bluetooth on my battery modules, but I could see the benefit for some. My one worry for a quick look is generally being able to disable over current protection is quite important for flying things. 

The other downside to these is that they use serial which means taking a serial port from another device. CAN and I2C would be better protocol choices as then you can have more batteries on the same bus.

I'm actively looking for companies and products in this space to integrate with ArduPilot. I received one positive response so far and Pomegrante Systems has been gracious enough to send me a few boards 

[Pomegrante Systems Power Module](https://p-systems.io/product/power_module)

Please let me know if you know of others?

geofrancis, 2021-12-04:
I use these for my custom ebike batteries and by battery trailer range extender, I havent had a single problem with them. as I said they are very popular with the ebike and scooter crowd due to them being very reliable and cost effective. sure canbus stuff would be nice but it just doesnt exist at the moment. neither do smbus devices outside a handful of very expensive premade batteries. 

here are a few other projects using them
https://github.com/kolins-cz/Smart-BMS-Bluetooth-ESP32
https://github.com/markandkymward/SMART-BMS-Bluetooth-Cell-Voltage-Monitor-HILTEC/
https://gist.github.com/JustGand/bdd6e1ae562c0dea850d3a023db42025

hendjoshsr71, 2021-12-04:
Awesome! Would you have recommendations for one or two poroducts (6s) and
any auxiliary products I would need for these?

geofrancis, 2021-12-04:
I personally have the 14s 30a board in my ebike battery and the 60a 6s for my trailer, there is an andoid/apple app for configuring them so I would get it with the bluetooth module so you can use the app to get up and running.

geofrancis, 2021-12-05:
If your planning on adding these I will order another for testing.

hendjoshsr71, 2021-12-05:
I am going to buy one. But no promises on any timeframe. It will take at least 4-6 weeks to get here. And not being one of the standard protocols it will take more time. (If the BMS was coming from from the battery cell/pack folks I know on AliExpress I'd be more inclined. But on AliExpress products come and go...)

amilcarlucas, 2021-12-06:
We are using a couple of Rotoye Batmon's, they are correctly integrated and work great as long as you do not use a CubeBlack flight controller.

The price is also ok, but yes you do need to e-mail them to get the barebone batmon without battery.

geofrancis, 2021-12-06:
> 

can you tell me what the price is?

geofrancis, 2022-01-31:
> We are using a couple of Rotoye Batmon's, they are correctly integrated and work great as long as you do not use a CubeBlack flight controller.
> 

those are just battery monitors, they cannot disconnect the battery incase of over/undervoltage or balance it.