ArduPilot/ardupilot #19446: ArduCopter: FRAME_CLASS/FRAME_TYPE = 0 when flashing new firmware

DonLakeFlyer, 2021-12-06:
I'm flashing onto Pixhawk 1. I tried Stable and Dev they both do the same thing. As far as I can remember this didn't used to be this way. This in turn causes QGC to connect to the board with minimal (not usable) UI since it has no vehicle type (like quad-rotor say). It used to default to quad-rotor X I think.  Then I tried to manually set FRAME_CLASS to 1 from the parameter editor I get back a PARAM_VALUE of 0 even though I just sent a 1. QGC needs a FRAME_CLASS/FRAME_TYPE to be set otherwise it is not going to work. I tried Heli and it works fine defaulting to a specific FRAME_CLASS/TYPE.

rmackay9, 2021-12-06:
Hi, thanks for the report.  I guess it is the FRAME_CLASS that is the issue.  It will appear as 0 by default on a new board so the user needs to then set it to 1 for Quad, 2 for Hexacopter, etc.

I don't think this behaviour has changed at all though..

DonLakeFlyer, 2021-12-07:
Hmm, I wasn't quite sure if this changed or not. Let me look some more. In general QGC would work much better if this was defaulted to something. Not quite sure what changed to make this work in current QGC.

DonLakeFlyer, 2021-12-07:
> Not quite sure what changed to make this work in current QGC.

Looks like this has been a problem for a while. Current Stable QGC is just as bad. Not sure if there is some way to work around it. The concept of understanding the "class" of vehicle type is fundamental to QGC + ArduPilot. This is because ArduPilot differs based on Copter/Plane/... and so forth. Even to the point of not being able to load something as simple as parameter meta data without knowing the vehicle class. Since the HEARTBEAT QGC gets in this case has no vehicle type set the majority of the UI is missing. I'm going to do some poking around but not sure if there is some way QGC can work around this.

DonLakeFlyer, 2021-12-07:
@rmackay9 So basically what I need to understand is that from the first HEARTBEAT that is an ArduCopter class vehicle. If ArduCopter is the only ArduPilot variants which doesn't set a FRAME_CLASS (and in turn I get HEARTBEAT.type == 0) after flash then I can make the assumption this is ArduCopter when I see that. Is that possible.

All of that said it still seems odd to me that as far as I can tell ArduCopter is the only variant that works this weewayk. Heli, Rover, Plane, ... all set FRAME_CLASS related stuff to a reasonable default. Even though that may not be correct in all cases and will need to be changed by the user.

DonLakeFlyer, 2021-12-07:
@rmackay9 If you look at the change I made and you think it will work then I've tested this and it makes QGC show useful UI in this case.

rmackay9, 2021-12-07:
@DonLakeFlyer, yes, that seems fine to me.  Sorry for the troubles.. I suppose we could modify AP to say that it is a quadcopter (just in the mavlink heatbeat) if the user hasn't specified yet..

hendjoshsr71, 2021-12-08:
@rmackay9  So not being as familiar with the history, I presume the reasoning behind not setting the FRAME_CLASS is to force users to do so? Is this behavior something that should be kept instead of defaulting to 1, for example?

So I tried playing with this in both Mission Planner and QGC on new board.  It seems like Mission Planner handles this by adding `FRAME_CLASS 0` as an undefined state. And having a button just for that.

DonLakeFlyer, 2021-12-08:
I merged the change. So all good now with QGC.

> I suppose we could modify AP to say that it is a quadcopter (just in the mavlink heatbeat) if the user hasn't specified yet..

Yes. That makes some sense. I'd still need the special case in QGC for now for back compat. But moving forward at some point if vehicle type was always set I could remove it further down the road.

drtrigon, 2021-12-25:
I can report that I setup a Matek H743 as Copter with 4.0.7 and had no issues - so FRAME_CLASS was not 0.

Later I tried to setup a Matek F405 with the 4.1.2 Copter using the same QGC version and run into this issue (FRAME_CLASS=0). I had to use APM Planner to fix it because QGC is not able to.

So this has either changed from 4.0 to 4.1 or is not consistent over all hardware or ArduPilot versions.

Another solution might be to setup QGC such that it just changes the FRAME_CLASS e.g. to 1 whenever it sees a 0. Or at least asks whether it should be changed. (It looks like 0 is actually never used as most of the GCS cannot handle it, right?)

drtrigon, 2022-01-04:
Today I continued with setting up the FC. `FRAME_CLASS = 0` happened about 3 times during this, e.g. after calibration of accelerometer it ask for reboot, after this boot I had again to set `FRAME_CLASS = 1` manually.

IamPete1, 2022-08-05:
Sounds like the QGC issue has been resolved. I don't think we sould change AP, forcing users to pick a frame class is the correct thing to do.

closed 2022-08-05 by IamPete1