ArduPilot/ardupilot #19452: AP_BatteryMonitor: remove use of double in mah calculation

rmackay9, 2021-12-07:
As pointed out by @WickedShell when reviewing PR https://github.com/ArduPilot/ardupilot/pull/19404 we probably don't need to cast to double in lines like below in AP_BattMonitor_UAVCAN.cpp.

`float mah = (float) ((double) _interim_state.current_amps * (double) dt * (double) 0.0000002778f);`

We should remove the doubles and make sure the accuracy of the output is not reduced.

shiv-tyagi, 2021-12-14:
@rmackay9 I would like to work on this issue. But I have one question, if we simply do to divide `_interim_state.current_amps * dt` by 3600000 we should surely get more precise results than multiplying by 0.0000002778f but we know that divisions are much costly than multiplication. So should I compromise cost for precision?

rmackay9, 2021-12-14:
@shiv-tyagi I wouldn't expect any difference in accuracy but maybe I'm misunderstanding something.  In general we should go for accuracy though.  By the way, when it comes to testing it would be good to check various values to ensure the calculations are the same before and after the change.

hendjoshsr71, 2021-12-28:
Closing. Thanks! @arBalasquide

closed 2021-12-28 by hendjoshsr71