ArduPilot/ardupilot #19455: Home altitude is not updating on SITL sim_vehicle.py

galdalali, 2021-12-07:
## Bug report

**Issue details**

If a user want to set the home altitude on runtime in SITL then there is a problem with updating the ground altitude since ground level updating once on startup where the home is - https://github.com/ArduPilot/ardupilot/blob/Copter-4.0.3/libraries/SITL/SIM_Aircraft.cpp#L91

For example, if a drone takes off from a top of an hill and then the user sets the home to be somewhere on the foothills and then he set RTL mode then the drone will disarm mid-air and report - `printf("Hit ground at %f m/s\n", velocity_ef.z);` since the `on_ground()` condition is incorrect.
 
**Version**
Copter-4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quadrotor X

**Hardware type**
SITL

peterbarker, 2021-12-09:
Unable to replicate on master:

[image]

Please replicate on master and provide a `.tlog`.  Or even better, an autotest :-)

hendjoshsr71, 2021-12-10:
Well you are using 4.03.... right? I'm quite sure we don't backport SITL changes.... and fixes. 9,000 commits ago....

galdalali, 2021-12-10:
Sad thing. i'll open PR..

peterbarker, 2024-07-31:
@galdalali did you open that PR?

Is this still an issue for you?

galdalali, 2024-07-31:
Don't know, stopped using it a while ago.

peterbarker, 2024-08-02:
> Don't know, stopped using it a while ago.

OK, a bit sad we didn't work this one out.  Closing now.

closed 2024-08-02 by peterbarker