ArduPilot/ardupilot #19479: Generic support for the PCA9685

pieniacy, 2021-12-10:
## Feature request

**Is your feature request related to a problem? Please describe.**
The problem is that we cannot use PCA9685 to extend output capabilities in a cheap and convenient way.
Another problem is that substituting inav build with ardupilot is less convenient - inav supports PCA9685 for servos.

**Describe the solution you'd like**
Port of `AP_HAL_Linux/RCOutput_PCA9685.*` to be a generic driver. In particular, it should be available under `AP_HAL_ChibiOS` to use it on fpv fc boards like Matek or Kakute F4 and F7.

**Describe alternatives you've considered**
As of now alternatives are sbus output over uart with sbus decoder or AP_Periph boards. In every case alternatives are more expensive (PCA9685 module costs < $5) and more resource consuming (blocking full uart istead of chaining I2C).

**Platform**
[x] All

**Additional context**
I have a working inav on Kakute F4 build. It uses a few servos through PCA9685 module. It would be nice to just switch to ArduPilot, which appears completely possible, with the exception of PCA9685. Apart from switching from inav, I believe this extension board is such a convenient way to connect a lot of servos that it may be useful in a lot of builds that use output restricted boards. And finally, it seems like a pretty basic feature that ArduPilot just should have and it feels strange to not support it on F4 and F7 boards when the support is already in the project, but only for Linux.

geofrancis, 2021-12-11:
I think they are working on expanding ardupilot to support 32 channels at the moment, I suspect that will need to be done before PCA9685 or other servo expansion like canbus servo could be added.

Rothammel, 2024-05-20:
+1
still no update 😭

pieniacy, 2024-05-20:
I don't think this will be worked on. Even INAV [dropped support](https://github.com/iNavFlight/inav/pull/8032) for this. I guess SBUS->PWM converters should be used instead.

closed 2024-05-20 by pieniacy