ArduPilot/ardupilot #19480: AP_Periph/DroneCAN: can not build any AP_Periph target - dronecan_dsdl.py fails

magicrub, 2021-12-10:
Native Ubuntu 20.04, WSL1, and WSL2 can not build AP_Periph targets /w new DroneCAN submodule. Plane v4.1.4 works fine (it's before the DroneCAN submodule was introduced):

Simple test:
```
./waf configure --board CubeOrange-periph
./waf AP_Periph
```
Result:
```
tom@DESKTOP-CSBSP2L:/mnt/c/Users/magic/Documents/src/ardupilot$ ./waf AP_Periph
Waf: Entering directory `/mnt/c/Users/magic/Documents/src/ardupilot/build/CubeOrange-periph'
Generating DSDLC for CANARD: /usr/bin/python /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/dronecan_dsdlc/dronecan_dsdlc.py -O /mnt/c/Users/magic/Documents/src/ardupilot/build/CubeOrange-periph/modules/DroneCAN/libcanard/dsdlc_generated /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/ardupilot /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/com /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/cuav /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/dronecan /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/mppt /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/uavcan
Failed to run:  /usr/bin/python /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/dronecan_dsdlc/dronecan_dsdlc.py -O /mnt/c/Users/magic/Documents/src/ardupilot/build/CubeOrange-periph/modules/DroneCAN/libcanard/dsdlc_generated /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/ardupilot /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/com /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/cuav /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/dronecan /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/mppt /mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/uavcan
expanding ardupilot.gnss.RelPosHeading
expanding uavcan.equipment.actuator.Status
expanding uavcan.equipment.ahrs.RawIMU
expanding uavcan.equipment.esc.RPMCommand
...
expanding uavcan.protocol.param.Value
expanding uavcan.tunnel.Broadcast
expanding uavcan.tunnel.Call
expanding uavcan.tunnel.Protocol

Traceback (most recent call last):
  File "/mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/dronecan_dsdlc/dronecan_dsdlc.py", line 182, in <module>
    assert not buildlist-builtlist, "%s not built" % (buildlist-builtlist,)
AssertionError: {'uavcan.protocol.param.GetSet', 'uavcan.tunnel.Broadcast', 'uavcan.protocol.dynamic_node_id.server.Entry', 'uavcan.protocol.AccessCommandShell', 'uavcan.equipment.power.CircuitStatus', 'uavcan.equipment.air_data.StaticPressure', 'uavcan.equipment.ice.reciprocating.CylinderStatus', 'uavcan.protocol.debug.KeyValue', 'uavcan.protocol.file.Read', 'uavcan.equipment.air_data.Sideslip', 'uavcan.protocol.CANIfaceStats', 'uavcan.protocol.param.NumericValue', 'uavcan.protocol.DataTypeKind', 'uavcan.equipment.esc.RawCommand', 'uavcan.equipment.camera_gimbal.GEOPOICommand', 'uavcan.protocol.Panic', 'uavcan.protocol.NodeStatus', 'uavcan.equipment.esc.RPMCommand', 'uavcan.equipment.air_data.IndicatedAirspeed', 'uavcan.equipment.air_data.AngleOfAttack', 'uavcan.protocol.file.EntryType', 'uavcan.protocol.file.BeginFirmwareUpdate', 'uavcan.equipment.air_data.TrueAirspeed', 'uavcan.protocol.dynamic_node_id.server.Discovery', 'ardupilot.equipment.trafficmonitor.TrafficReport', 'uavcan.protocol.file.GetInfo', 'uavcan.equipment.hardpoint.Command', 'uavcan.protocol.param.Empty', 'uavcan.protocol.dynamic_node_id.server.AppendEntries', 'ardupilot.indication.NotifyState', 'uavcan.equipment.actuator.Status', 'uavcan.Timestamp', 'uavcan.equipment.indication.BeepCommand', 'uavcan.protocol.GlobalTimeSync', 'uavcan.equipment.gnss.ECEFPositionVelocity', 'uavcan.equipment.hardpoint.Status', 'uavcan.equipment.ahrs.MagneticFieldStrength2', 'mppt.Stream', 'uavcan.equipment.camera_gimbal.Status', 'uavcan.equipment.ahrs.MagneticFieldStrength', 'ardupilot.gnss.Status', 'uavcan.protocol.RestartNode', 'uavcan.equipment.gnss.RTCMStream', 'uavcan.equipment.ice.reciprocating.Status', 'uavcan.protocol.HardwareVersion', 'uavcan.equipment.device.Temperature', 'ardupilot.equipment.power.BatteryInfoAux', 'uavcan.navigation.GlobalNavigationSolution', 'uavcan.equipment.air_data.RawAirData', 'uavcan.protocol.debug.LogLevel', 'ardupilot.gnss.RelPosHeading', 'uavcan.protocol.param.Value', 'uavcan.protocol.GetNodeInfo', 'uavcan.tunnel.Protocol', 'uavcan.equipment.power.PrimaryPowerSupplyStatus', 'uavcan.protocol.dynamic_node_id.Allocation', 'uavcan.equipment.camera_gimbal.Mode', 'uavcan.equipment.actuator.Command', 'uavcan.protocol.file.Path', 'ardupilot.indication.Button', 'mppt.OutputEnable', 'uavcan.equipment.indication.RGB565', 'uavcan.equipment.esc.Status', 'uavcan.equipment.safety.ArmingStatus', 'uavcan.equipment.indication.LightsCommand', 'uavcan.equipment.camera_gimbal.AngularCommand', 'uavcan.protocol.file.Error', 'uavcan.equipment.ice.FuelTankStatus', 'uavcan.protocol.GetTransportStats', 'cuav.equipment.power.CBAT', 'uavcan.equipment.gnss.Fix2', 'uavcan.equipment.indication.SingleLightCommand', 'uavcan.protocol.file.Write', 'ardupilot.gnss.MovingBaselineData', 'uavcan.equipment.air_data.StaticTemperature', 'dronecan.sensors.hygrometer.Hygrometer', 'uavcan.protocol.param.ExecuteOpcode', 'ardupilot.indication.SafetyState', 'uavcan.protocol.file.GetDirectoryEntryInfo', 'uavcan.equipment.ahrs.RawIMU', 'ardupilot.gnss.Heading', 'uavcan.protocol.GetDataTypeInfo', 'uavcan.protocol.SoftwareVersion', 'uavcan.CoarseOrientation', 'com.hex.equipment.flow.Measurement', 'uavcan.tunnel.Call', 'uavcan.equipment.gnss.Fix', 'uavcan.equipment.range_sensor.Measurement', 'uavcan.protocol.enumeration.Indication', 'uavcan.equipment.ahrs.Solution', 'uavcan.equipment.actuator.ArrayCommand', 'uavcan.protocol.enumeration.Begin', 'uavcan.equipment.power.BatteryInfo', 'uavcan.protocol.debug.LogMessage', 'uavcan.protocol.file.Delete', 'uavcan.protocol.dynamic_node_id.server.RequestVote', 'uavcan.equipment.gnss.Auxiliary'} not built

Traceback (most recent call last):
  File "/mnt/c/Users/magic/Documents/src/ardupilot/modules/waf/waflib/Scripting.py", line 158, in waf_entry_point
    run_commands()
  File "/mnt/c/Users/magic/Documents/src/ardupilot/modules/waf/waflib/Scripting.py", line 251, in run_commands
    ctx = run_command(cmd_name)
  File "/mnt/c/Users/magic/Documents/src/ardupilot/modules/waf/waflib/Scripting.py", line 235, in run_command
    ctx.execute()
  File "/mnt/c/Users/magic/Documents/src/ardupilot/Tools/ardupilotwaf/ardupilotwaf.py", line 201, in execute
    return execute_method(self)
  File "/mnt/c/Users/magic/Documents/src/ardupilot/modules/waf/waflib/Scripting.py", line 573, in execute
    return execute_method(self)
  File "/mnt/c/Users/magic/Documents/src/ardupilot/modules/waf/waflib/Build.py", line 231, in execute
    self.execute_build()
  File "/mnt/c/Users/magic/Documents/src/ardupilot/modules/waf/waflib/Build.py", line 244, in execute_build
    self.recurse([self.run_dir])
  File "/mnt/c/Users/magic/Documents/src/ardupilot/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/mnt/c/Users/magic/Documents/src/ardupilot/wscript", line 759, in build
    _load_pre_build(bld)
  File "/mnt/c/Users/magic/Documents/src/ardupilot/wscript", line 741, in _load_pre_build
    generate_dronecan_dsdlc(bld)
  File "/mnt/c/Users/magic/Documents/src/ardupilot/wscript", line 530, in generate_dronecan_dsdlc
    raise RuntimeError('Failed to generate DSDL C bindings')
RuntimeError: Failed to generate DSDL C bindings
```

hendjoshsr71, 2021-12-10:
This is on master? Unless I misunderstood? My periph knowledge is lacking but your commands work for me....
```bash
./waf configure --board CubeOrange-periph
./waf AP_Periph > waf_ap_periph_build.txt 2>&1
```
[waf_ap_periph_build.txt](https://github.com/ArduPilot/ardupilot/files/7695873/waf_ap_periph_build.txt)

magicrub, 2021-12-10:
well, there seems to be something funny going on with my setup. I've spun up a fresh Ubuntu 18.04 and it works fine so I'll close this and keep digging

closed 2021-12-10 by magicrub

hendjoshsr71, 2021-12-10:
I find i have to `./waf distclean` and `./Tools/gittools/submodule_update.sh` all the time ... but I'm sure you tried that ;(

magicrub, 2021-12-10:
yup, for sure. I think this is certainly my root cause, thinking it's actually a waf configure problem

Mine:
```Generating DSDLC for CANARD:
/usr/bin/python /
mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/dronecan_dsdlc/dronecan_dsdlc.py -O 
/mnt/c/Users/magic/Documents/src/ardupilot/build/CubeOrange-periph/modules/DroneCAN/libcanard/dsdlc_generated 
/mnt/c/Users/magic/Documents/src/ardupilot/libraries/AP_UAVCAN/dsdl/ardupilot 
/mnt/c/Users/magic/Documents/src/ardupilot/libraries/AP_UAVCAN/dsdl/com 
/mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/ardupilot 
/mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/com 
/mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/cuav 
/mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/dronecan 
/mnt/c/Users/magic/Documents/src/ardupilot/modules/DroneCAN/DSDL/uavcan
```

Yours:
```Generating DSDLC for CANARD: 
/usr/bin/python 
/home/veritas/ardupilot/modules/DroneCAN/dronecan_dsdlc/dronecan_dsdlc.py -O 
/home/veritas/ardupilot/build/CubeOrange-periph/modules/DroneCAN/libcanard/dsdlc_generated 
/home/veritas/ardupilot/modules/DroneCAN/DSDL/ardupilot 
/home/veritas/ardupilot/modules/DroneCAN/DSDL/com 
/home/veritas/ardupilot/modules/DroneCAN/DSDL/cuav 
/home/veritas/ardupilot/modules/DroneCAN/DSDL/dronecan 
/home/veritas/ardupilot/modules/DroneCAN/DSDL/mppt 
/home/veritas/ardupilot/modules/DroneCAN/DSDL/uavcan
```

It seems to be slipping in ```/ardupilot/libraries/AP_UAVCAN/dsdl/```

magicrub, 2021-12-11:
@peterbarker and I tracked it down. Problem with a python module. There will be a DroneCAN PR soon to address it.