ArduPilot/ardupilot #19493: BTN_FUNC maintenance nightmare

Hwurzburg, 2021-12-11:
The aux function feature of buttons presents a maintenance nightmare currently.....every addition/change in an RCx_OPTION must be reflected in the four BTN_FUNCx param metadata fields.....they are always forgotten

possible solutions:
1. eliminate those declarations in BTN_FUNCx and simply refer to RC1_OPTIONS , ie "@Value: For a list of auxiliary function values see the RC1_OPTION parameter list"
2. create a script that generates/duplicates the RC1_OPTION @Value metadata lists in each of the four BTN_FUNCx params at build time
3.  ?????

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2021-12-11:
We have the same issue with all the various analog and digital pin input functions. Some sort of shared scheme would be useful there too.

rmackay9, 2021-12-14:
On the dev call today @tridge and @peterbarker suggested they might extend the parameter parser to be a two-pass parser and support markup like below

```
@VALUESSAVE: RCOPTION
@Values: $(RCOPTION)
```

closed 2021-12-14 by tridge