ArduPilot/ardupilot #19519: Lat/Lon/Altitude in MAV_CMD_DO_MOUNT_CONTROL not decoded properly

marshallharris, 2021-12-14:
## Bug report

**Issue details**

In order to command the gimbal to point at target specified by a given coordinate in a non-mission flight mode (not Auto), the MAVLink protocol indicates I should send a [MAV_CMD_DO_MOUNT_CONTROL](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_MOUNT_CONTROL) in a COMMAND_LONG with the following parameters.
Param 1, 2, 3: 0 (Pitch, Roll and Yaw are not needed)
Param 4: Target Altitude
Param 5: Target Latitude
Param 6: Target Longitude
Param 7: 4 (MAV_MOUNT_MODE_GPS_POINT)

However, when sending this command to the SITL, it crashes with the error `Should not be called on invalid location`

Further inspection revealed that the issue is that [the function](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Backend.cpp#L54)

`AP_Mount_Backend::control(int32_t pitch_or_lat, int32_t roll_or_lon, int32_t yaw_or_alt, MAV_MOUNT_MODE mount_mode)` 

expects the first three arguments to be either pitch/lat, roll/lon and yaw/alt and the [calling function](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount.cpp#L586) only supplies the arguments with param1, 2 and 3 of the `mavlink_command_long_t` packet. Even when the mode is MAV_MOUNT_MODE_GPS_POINT.

```
MAV_RESULT AP_Mount::handle_command_do_mount_control(const mavlink_command_long_t &packet)
{
    if (!check_primary()) {
        return MAV_RESULT_FAILED;
    }

    // send message to backend
    _backends[_primary]->control(packet.param1, packet.param2, packet.param3, (MAV_MOUNT_MODE) packet.param7);

    return MAV_RESULT_ACCEPTED;
}
```

The SITL is crashing because it thinks the target location to be invalid (0 for lat, lon and alt) when it not looking at the correct fields for those values. I suspect this interpretation of MAV_CMD_DO_MOUNT_CONFIGURE followed [the transition from MOUNT_CONTROL](https://github.com/ArduPilot/ardupilot/pull/2696).

I understand that I can set a location target by supplying lat, lon and alt as Parameters 1, 2 and 3 in my command, but this departure from the MAVLink protocol isn't documented anywhere I could find and I am unclear as to why the departure was made at all.

Thanks

**Version**
4.1.0

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad

**Hardware type**
SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

amilcarlucas, 2021-12-15:
We use our own Mavlink protocol fork, this is one of the few messages that is different from the upstream mavlink repository.

PR`s that improve it somehow are welcome.

marshallharris, 2021-12-15:
Ok, thank you!

arBalasquide, 2021-12-16:
@amilcarlucas Hi, is there a reason as to why ArduPilot has would need a different MAV_CMD_DO_MOUNT_CONTROL definition?

amilcarlucas, 2021-12-17:
Yes, the upstream version was very broken, and kept changing. We adopted a more long term stable strategy.

rmackay9, 2022-12-19:
I've fixed the two problems mentioned above over the past few months so I think I will close this.  I'm very happy to re-open it though if @marshallharris says the issue still exist.

closed 2022-12-19 by rmackay9