ArduPilot/ardupilot #19525: Rover/Boat: add wave height estimate (lua script?)

rmackay9, 2021-12-15:
It could be useful for users if AP could estimate the current wave height.  Perhaps this could be done by maintaining an EKF altitude bias estimate (assuming that over a long period of time the altitude must not be climbing or descending) and then compare this to the current EKF altitude estimate looking for the peak difference over the past minute.

IamPete1, 2021-12-15:
I added the wave simulation with this in mind. I was thinking to integrate it out of the roll / pitch angles. But as you say it might be that the EKF already does that internally. 

I was hoping we might be able to get period, amplitude and a heading for waves.

geofrancis, 2021-12-15:
I think this might have to be looked at in 2 ways because waves aren't just like the classic waves there are also swells, these are essentially much longer lower frequency waves that lift ships rather than hitting them. 

In randys video https://youtu.be/V8N3lA-20fs?t=581 you can kind of see what im talking about, if you had a sensor on the very bow of the boat and recorded its up and down movement you would record the energy of the surface waves generated by the wind at around 2-10hz depending on the speed of the boat but if you looked at something like the barometric or a RTK altimeter you would see much slower oscillations following the large swells probably closer 0.5hz

geofrancis, 2021-12-15:
on a side note, your in japan, couldnt something like be used for tsunami detection? park a solar boat out at sea with a RTK gps monitoring its altitude.

urenaf122, 2023-01-09:
the best way, in my opinion, to counteract the high waves tilting the boat is by placing spring hinged pontoon like shown in this utube video

https://youtu.be/qvL3XvpyFkY