ArduPilot/ardupilot #19532: AP_CANManager: remove MPPT_PacketDigital

rmackay9, 2021-12-16:
As a follow-up to issue https://github.com/ArduPilot/ardupilot/issues/19453 and PR https://github.com/ArduPilot/ardupilot/pull/19458 we should also remove, "20:MPPT" from the BATTx_MONITOR parameter description.

Also we should remove AP_CANManager's "AP_CANManager::Driver_Type_MPPT_PacketDigital" including AP_CANManager's parameter description for, "9:PacketDigital".

MohdImran001, 2021-12-16:
Hi @rmackay9, I would like to work on this issue.

hendjoshsr71, 2021-12-16:
Just a note @shiv-tyagi was part way along this one, but I'm not sure if he wants to finish it?

For example for CANManager here:
https://github.com/ArduPilot/ardupilot/blob/317d32574cf77948691d1e5ddc24c89d6281f319/libraries/AP_CANManager/AP_CANManager.h#L63-L65

@rmackay9  Could we get your thoughts on the below? 
There are three removal options I see:
1.) Remove `9:PacketDigital` and just skip the number 9 (which would be odd for users)
2.) Remove it and convert the other parameter
3.) Remove it and dont convert because the parameters after it never made it into a stable release?

The same options should be available AP_BattMonitor:
https://github.com/ArduPilot/ardupilot/blob/317d32574cf77948691d1e5ddc24c89d6281f319/libraries/AP_BattMonitor/AP_BattMonitor_Params.cpp#L16

rmackay9, 2021-12-16:
@hendjoshsr71,

This driver was actually included in 4.1 but I don't think there were more than 1 or two users and @magicrub would know them all.

I think we should remove all the CAN code for MPPT_PacketDigital/9 option although I'm less sure about what to do about the enum.  My feeling is we should remove it too and leave a comment in it's place saying something like, "// 9 was MPPT_PAcketDigital".

rmackay9, 2021-12-16:
Whomever creates a PR first wins!

hendjoshsr71, 2021-12-20:
Closing. Thanks @shiv-tyagi !

closed 2021-12-20 by hendjoshsr71