ArduPilot/ardupilot #19535: Rover: maximum steering deflection is reduced at higher speeds on Ackermann vehicles

rmackay9, 2021-12-16:
The steering input is constrained (in some modes at least) to +-4500 before being passed into the motors library.  Within the motors library it then scales this down further (on Ackermann steering vehicles) if the vehicle's velocity is above MOT_SPD_SCA_BASE.

This means that the maximum movement of the steering is reduced at higher speeds.  Instead we should perform the scaling before the limiting so that the maximum range is always available.

Thanks very much to mroberts for uncovering this issue here https://discuss.ardupilot.org/t/understanding-history-and-logic-of-mot-spd-sca-base/79450/11

mikerob, 2022-01-04:
Note that to successfully test in sim, you will need to bump up the turning circle of the sim Rover to 10m or so in order for the FF gain to be high enough to show the effect.

rmackay9, 2022-02-15:
This is now resolved in master so closing.

closed 2022-02-15 by rmackay9