ArduPilot/ardupilot #19537: Support DS18b20 temperature sensor

geofrancis, 2021-12-16:
As far as i can tell currently the only way to get an external temperature probes is to use an external barometer, this isn't ideal for monitoring motors and electronics. what would be is the DS18b20 temperature probe, they are accurate and cost pennies. 
it uses a single wire serial protocol, each probe has a unique id, they just need a pull up resistor on the data wire.

https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf
https://www.maximintegrated.com/en/products/sensors/DS18B20.html
https://cdn.sparkfun.com/datasheets/Sensors/Temp/DS18B20.pdf

https://github.com/hepingood/ds18b20
https://github.com/matmunk/DS18B20

sq6emm, 2022-09-25:
I'd love to use it also, as I need to monitor my DC engines temperatures in huge EV only grass mover - it can cut 1 meter width at a time...;) So temperature is very important!;)

geofrancis, 2022-09-25:
> I'd love to use it also, as I need to monitor my DC engines temperatures in huge EV only grass mover - it can cut 1 meter width at a time...;) So temperature is very important!;) Or maybe any other temp sensors are currently supported?;)

I forgot about this issue, yuri_rage has made an arduino sensor adapter and there is an example using the DS18b20 sensor. 

https://discuss.ardupilot.org/t/unsupported-sensors-try-arduino-lua/86814

closed 2022-09-25 by geofrancis