ArduPilot/ardupilot #19566: Add json variant of Parametr Definitions to autotest.ardupilot.org/

wucke13, 2021-12-18:
## Feature request

**Is your feature request related to a problem? Please describe.**
I'd like to build a GCS application. The XML of the parameter definition is _not optimal_, both due to the verbosity of XML and the actual structuring of the data. The JSON variant of the parameters is way nicer for parsing. Now https://ardupilot.org/dev/docs/gcs-resources.html points GCS developers towards the https://autotest.ardupilot.org/Parameters/ArduCopter/ page, which lists the parameter lists in various formats, __but not JSON__. At least from my perspective, JSON would probably be the best format for GCS developers, since it by now is the most ubiquitous data-format.

**Describe the solution you'd like**
Generate and provide JSON parameter definition files on https://autotest.ardupilot.org

**Describe alternatives you've considered**
We could ask every potential GCS developer to generate the data locally, on demand. This is suboptimal: In my GCS I want to allow the user to import an updated parameter definitions file, possibly directly via via URL. The GCS then downloads the JSON from the URL, stores it locally for caching and the user has the newest parameter definitions.

**Platform**
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2021-12-20:
I've created a patch to fix our json output.  I'm not actually 100% sure this will fix the entirely missing file, but I've attached what *should* be appearing after that patch is applied.
[apm.pdef.json.txt](https://github.com/ArduPilot/ardupilot/files/7742073/apm.pdef.json.txt)

(renamed to add .txt because github...)

peterbarker, 2021-12-20:
... I've now pushed up a patch which will populate the directory on the autotest server with the JSON output (@patrickelectric )

wucke13, 2021-12-20:
@peterbarker Thank you! When will it be live?

peterbarker, 2021-12-20:
> @peterbarker Thank you! When will it be live?

A few hours after that PR has been merged.

It would be nice to know the format is actually suitable for your purposes before I merge it - have you had a chance to look at the output?  I'm not sure how much flexibility I have to change it - I think @patrickelectric might be using it in its current form.

wucke13, 2021-12-20:
I think the format is decent, definitely way better than the XML (which is just super inconsistent in many ways). Of course I'd like to have all vehicle types combined in a signle JSON. The current format already allows to make quite informative GUIs - I'll attach a screenshot of my early progress parameter editor.
[image]

closed 2021-12-21 by tridge

peterbarker, 2021-12-21:
@wucke13 most people only want 1 or 2 vehicles, so download times are a bit of an issue.  Can't reuse library data as they're subtly different on different vehicles, so it really is just the sum of all the other files anyway.

In any case, you should find that file starts to appear in a little while, since we've merged it...