ArduPilot/ardupilot #19594: Yaw turning direction only clockwise when using MAV_CMD_DO_MOUNT_CONTROL with 2-axis gimbal

hhackbarth, 2021-12-22:
## Bug report

**Issue details**

Setting gimbal yaw on 2-axis gimbals using MAV_CMD_DO_MOUNT_CONTROL results in the copter changing it's heading to the yaw angle desired for the gimbal as discussed e.g. in issue #15699 or https://discuss.ardupilot.org/t/mission-do-mount-control-impacts-yaw-of-copter-bug-or-feature/37497

In ArduCopter versions up to somewhere in the 4.1.0-dev era, the direction for the yaw turn of the drone was determined automatically to minimize the turn. E.g. if the copter's heading was 90 degree (east) and the gimbal yaw was set to 60 degree with MAV_CMD_DO_MOUNT_CONTROL, the copter turned counterclockwise (ccw) by 30 degree. If yaw was set to 120 degree, the copter turned clockwise (cw) by 30 degree.

Since some time, when using ArduCopter 4.2.0-dev in SITL, the drone **always** turns clockwise. So, if it has a heading of 90 degree and gimbal yaw is set to 85 degree using MAV_CMD_DO_MOUNT_CONTROL, the drone makes almost a full turn clockwise (by 355 degree) instead of turning only 5 degrees ccw.

Unfortunately the gimbal mount_control command has no parameter that allows to control the direction of the rotation.

There may be a dependency to the behaviour of command MAV_CMD_CONDITION_YAW which also seems to have changed somewhere between 4.1.0-dev and 4.2.0-dev. In the past, the direction of the yaw turn was only controlled by the direction parameter (param3) when param4 was set to 1 (relative). In case of param4 = 0 (absolute), the yaw direction change was automatically choosen depending on the current heading and the target heading. In AC 4.2.0-dev, also for absolute target headings the direction is now controlled by param3.

The new behaviour of the MAV_CMD_DO_MOUNT_CONTROL can be very annoying. If you only want to change the gimbal pitch, you have to determine the drone's heading and use it as yaw for the gimbal rotation command. If the drone's heading has changed a little bit in clockwise direction (e.g. from 90 degree to 91 degree) and you set the new pitch still using 90 degree as yaw, it results in a 359 degree turn of the drone to set the new target angel for the gimbal...

I checked this behaviour with SITL (configured as quad frame with a gimbal MNT_TYPE set to 3 (Alexmos)) but I assume, it will be the same also for other mount types when the gimbal only has 2 axis (has_pan() is set to 0 / false).

**Version**
4.2.0-dev (SITL)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
SITL (gimbal MNT_TYPE = 3 (AlexMos))

**Logs**

IamPete1, 2022-09-04:
@rmackay9 Lots of mount improvements recently, I wonder if this has been fixed?

rmackay9, 2022-09-05:
I've tested this quite thoroughly and can't reproduce the problem so I think it has been fixed.

closed 2022-09-05 by rmackay9