ArduPilot/ardupilot #19596: GCS_MAVLink: incorrect interpretation of MAV_CMD_DO_REPEAT_SERVO's cycle time field

rmackay9, 2021-12-23:
According to the MAVLink spec for the MAV_CMD_DO_REPEAT_SERVO mavlink message, "param4" ("Cycle time") should be in seconds but while testing this command I found that we interpret it in milliseconds.

Note that I'm not talking about an Auto mode mission command but rather an "immediate" command sent from the GCS or companion computer.  Here is how to reproduce the issue in SITL:

- start SITL with the vehicle of your choice
- module load message
- module load graph
- graph SERVO_OUTPUT_RAW.servo8_raw
- message COMMAND_LONG 0 0 184 0 8 1200 3 **1000** 0 0 0

You should see the servo output oscillate at 1hz but really the "1000" should be "1" if we are interpreting the mavlink as per the spec.

muramura, 2021-12-24:
The same goes for do_repeat_relay.
https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPEAT_RELAY

tridge, 2021-12-27:
looks like px4 doesn't support DO_REPEAT_SERVO, so we can propose changing the xml