ArduPilot/ardupilot #19597: dis-Compatibility to documentation

galdalali, 2021-12-23:
The `MAV_MOUNT_MODE_MAVLINK_TARGETING` should receive it's target in degrees according to [documentation](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_MOUNT_CONTROL).
The ArduPilot [code ](https://github.com/ArduPilot/ardupilot/blob/312604f57773cce100bd8e5e6b4224cd8f49fe18/libraries/AP_Mount/AP_Mount_Backend.cpp#L67)from the other hand expect to receive it in centi-degrees

amilcarlucas, 2022-01-03:
We are using our own [mavlink fork](https://github.com/ArduPilot/mavlink/) and [as you can see](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L1049) use a different definition for that message.

Pull request correcting the documentation are very welcome.

IamPete1, 2022-09-04:
closed by https://github.com/ArduPilot/ardupilot/pull/21530

closed 2022-09-04 by IamPete1