ArduPilot/ardupilot #19607: Position target set in GUIDED mode is slightly corrupted by Arducopter source code

ardugitter, 2021-12-24:
## Bug report

**Issue details**

1. Companion computer uses MAVLINK_MSG_ID_**SET**_POSITION_TARGET_GLOBAL_INT to set these:

```
int32_t lat_int
int32_t lon_int
float alt
```

2. Companion computer requests MAVLINK_MSG_ID_POSITION_TARGET_GLOBAL_INT messages from Arducopter. The received values of lat_int, lon_int and alt are often not precisely the same as those sent to Arducopter in 1. above. 

Here is an example of the lat_int, lon_int and alt as set with SET_POSITION_TARGET:
2897889 57051 29.790001
Here is an example of what is received back:
2897888 57051 29.789999

**Version**
_What version was the issue encountered with_

Arducopter 4.1.1

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Pixhawk 1

The problem occurs because Arducopter converts the precise binary values that are set by the SET_POSITION_TARGET message before saving them for use later when they are sent back to the companion computer as part of regular POSITION_TARGET messages.

We see in ./ArduCopter/GCS_Mavlink.cpp the apparent change of accuracy of **alt**:

loc = {packet.lat_int, packet.lon_int, int32_t(packet.alt*100), frame};

I understand that loc needs an altitude in cm, but the fact is the position target is a fixed binary value that should be sent back to the companion computer *exactly* as it was requested IMO or how can the companion computer check that it has set a position target correctly?

We see in location.cpp these lines which affect accuracy and cause the "off by 1" error in **lat_int** and **lon_int**:

vec_ne.x = (lat-ekf_origin.lat) * LATLON_TO_CM;
vec_ne.y = diff_longitude(lng,ekf_origin.lng) * LATLON_TO_CM * longitude_scale((lat+ekf_origin.lat)/2);

Please would it be possible to return the precise position target set by the companion computer?

There is no other way to confirm that a position target has been set (i.e. there is no ACK).

Thanks for reading.

khancyr, 2021-12-24:
hello,

What is send is not exactly what was requested, it is what the drone is targetting. According to our current accuracy on float value, those are two position are the same (float are limited to 1e-5 accurracy)

On you companion computer you should use the WP_RADIUS parameter as the accuracy to validate that the position is right as that is what ArduPilot does internally.

lthall, 2021-12-25:
These messages are something I would like to define what our formal interpretation is.

POSITION_TARGET_GLOBAL_INT ( #87 )
Reports the current commanded vehicle position, velocity, and acceleration as specified by the autopilot. This should match the commands sent in SET_POSITION_TARGET_GLOBAL_INT if the vehicle is being controlled this way.

The description here is self contradictory. We can not simultaneously return "current commanded vehicle position, velocity, and acceleration" and the previous command sent using SET_POSITION_TARGET_GLOBAL_INT. For example, if both position and velocity is commanded the "current commanded vehicle position" changes over time based on the "current commanded vehicle velocity". The only time they will match is at the moment the original command was received.

My personal interpretation is:
Reports the current commanded vehicle position, velocity, and acceleration as specified by the autopilot. This should **be consistent with** the commands sent in SET_POSITION_TARGET_GLOBAL_INT if the vehicle is being controlled this way.

So this would be the instantaneous targets being fed to the the position controller.

Based on my interpretation a companion computer would need to handle things like floating point accuracy issues and integration differences like I described above.

ardugitter, 2021-12-25:
> What is send is not exactly what was requested, it is what the drone is targetting. According to our current accuracy on float value, those are two position are the same (float are limited to 1e-5 [...]

Okay, I suppose I will make do with the slightly reduced accuracy. It is easy to program for. But I *do* need a way to confirm that Arducopter has received the position target.

I have WPNAV_RADIUS set to 200. That seems quite big - a 2m radius!

Thanks for confirming these things to me. :-)

ardugitter, 2021-12-25:
> Reports the current commanded vehicle position, velocity, and acceleration as specified by the autopilot. This should match the commands sent in SET_POSITION_TARGET_GLOBAL_INT if the vehicle is being controlled this way.

Having read the Mavlink description you provided, the word "match" usually means "is the same as". If Arducopter is changing the set values in any way, it is not adhering to the Mavlink spec IMO.

I don't think I follow your post exactly, but the way things work with position targets at present seems logical: you use SET_POSITION_TARGET to tell Arducopter where you want to fly, then Arducopter tells you the location that it is flying to using POSITION_TARGET. This POSITION_TARGET does not change until you specify a new location to fly to.

lthall, 2021-12-26:
This is true if you are using the most basic control case as you are here. However if you are providing both position and velocity you are stating you want to be at this position doing this velocity. So you are commanding the target position to move over time. In that case the POSITION_TARGET_GLOBAL_INT will only match what you sent at the time the message was received because one loop later the target position will move based on your commanded velocity.

If we interoperated the message on the second sentence rather than the first then we would need to provide the target position and velocity with a time stamp in the past matching when you sent the message. But to do this we would need to completly ignore the first sentence in that specification "Reports the **_current_** commanded vehicle position, velocity, and acceleration as specified by the autopilot."

My interpretation is the first sentence is the specification and the second is providing an example but the example is only accurate in some cases (at the moment the message is receive or when velocity and acceleration are commanded as zero).

ardugitter, 2021-12-26:
Okay, so we agree that in the specific case of just commanding location, Arducopter should send back the matched position target as requested.

> So you are commanding the target position to move over time. In that case the POSITION_TARGET_GLOBAL_INT will only match what you sent at the time the message was received because one loop later the target position will move based on your commanded velocity.

I didn't realise that the drone would keep updating its target if you demand a velocity and location. I just tried setting a dual velocity/location target using SITL and I did not receive any position targets at all in this scenario. That said, the drone did fly towards the position target that I requested.

I get position targets fine when specifying just location (no velocity).

lthall, 2021-12-26:
> I get position targets fine when specifying just location (no velocity).

I just checked and you are correct. send_position_target_global_int() only returns the position while send_position_target_local_ned() returns the position, velocity and acceleration like I described. It looks like send_position_target_global_int only reports when the waypoint controller is being used and that is position only. I will have to add that to my list of things to look at.

> I have WPNAV_RADIUS set to 200. That seems quite big - a 2m radius!

You should be able to just check the returned value is +- 1 from your commanded value.

hendjoshsr71, 2021-12-26:
As a note here is my PR to fill out this message: https://github.com/ArduPilot/ardupilot/pull/18580
It on my list of something we should get into 4.2

ardugitter, 2021-12-26:
> You should be able to just check the returned value is +- 1 from your commanded value.

Okay, so it is more like a diameter value than radius? Sorry to be pedantic. :-)

lthall, 2021-12-26:
>It is on my list of something we should get into 4.2

Nice. Lets make it happen.

>Okay, so it is more like a diameter value than radius? Sorry to be pedantic. :-)

Ahh, no, it is a cube...... +-1 on each axis. :) I can out pedantic anybody.

ardugitter, 2021-12-27:
> Ahh, no, it is a cube...... +-1 on each axis. :) I can out pedantic anybody.

Need a pull request to change "WPNAV_RADIUS" to "WPNAV_HALFCUBESIZE" :-D

lthall, 2021-12-27:
hahah, I was talking about the implementation of your check, not WP_Radius. WP_Radius is a sphere (and a radius).