ArduPilot/ardupilot #19609: Free Flash (B) in BUILD SUMMARY is not correct when all vehicles are built

Jaaaky, 2021-12-24:
When I simply run `./waf` without specifying a vehicle type Free Flash (B) is not correctly calculated for each. Single value is displayed for all

For example I get:
```
Target               Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
--------------------------------------------------------------------------------------
bin/antennatracker     991544      1596   195216                993140         1087628
bin/arducopter        1297756      1600  -111012               1299356         1087628
bin/arducopter-heli   1273960      1600   -87216               1275560         1087628
bin/arduplane         1192108      1588    -5336               1193696         1087628
bin/ardusub           1191476      1616    -4752               1193092         1087628
bin/blimp              920780      1592   265980                922372         1087628
bin/ardurover         1161164      1588    25608               1162752         1087628

```

Jaaaky, 2021-12-24:
BSS too is incorrectly displayed "negative numbers???"

khancyr, 2021-12-24:
Nice one. What target did  you build? 
We got some issue with the crashdump for binary size calculus. It looks like we miss a case on some target or we got build issue on those target

Jaaaky, 2021-12-24:
Pixhawk1 .. I think it's applicable to all ChibiOS boards.

hendjoshsr71, 2021-12-24:
Is this on MacOS?

Known issue from before my PR where these values where incorrect on master.
My PR to fix the crash debug issue made this more pronounced.

hendjoshsr71, 2021-12-25:
Replicated on WSL.  Will fix! Sorry about that!

closed 2021-12-27 by tridge