ArduPilot/ardupilot #19616: MacOS: Arm Builds for same Board Type and GIT Hash are Different Sizes

hendjoshsr71, 2021-12-28:
@shiv-tyagi was testing some changes for me on MacOS  and noticed that the build size between his MacOS and Linux were coming out with different sizes by about 2KB on the the same GIT hash.

We checked his `./waf configure --board=CubeOrange` and this is his configuration output for mac:
[message.txt](https://github.com/ArduPilot/ardupilot/files/7783059/message.txt)

He is using: gcc-arm-none-eabi-10-2020-q4-major , g++ 10.2.1 

Here are the summaries from the two different .apj's

amilcarlucas, 2022-01-03:
What gcc-arm version are you using?

khancyr, 2022-01-03:
what is the python version ?

hendjoshsr71, 2022-01-03:
@shiv-tyagi Would please try this out again for us? 

And upload the text from `./waf configure --board=CubeOrange` for both your linux and MacOS. 

Also the .apjs from each for say copter.

shiv-tyagi, 2022-01-04:
[Mac.zip](https://github.com/ArduPilot/ardupilot/files/7805209/Mac.zip)

hendjoshsr71, 2022-01-06:
@shiv-tyagi Thanks again for testing the MacOS stuff!

**Ubuntu**
You are using g++ 9.2.1 for your arm compiler on Ubuntu. Likely the preinstalled version? You need to be sure to use the project supplied compiler here, gcc 10.2.1, https://firmware.ardupilot.org/Tools/STM32-tools/

 I think we can close this as your Mac APJ is very close, within  around 180 bytes, to when I built it on WSL. We could figure out why is different because I believe it should be exact? But it is a lot less worrying that the 2+ KB difference which is likely due to your complier difference,

closed 2022-01-06 by hendjoshsr71