ArduPilot/ardupilot #19630: cannot read imu data via mavros

tiralonghipol, 2021-12-29:
## Bug report

**Issue details**
no data when subscribing to /mavros/imu/data or /mavros/imu/data_raw

I recently got the iFlight Beast H7 55A AIO and I succesfully flashed ardupilot on it
but it seems like I am unable to read data via mavros

**Version**
_What version was the issue encountered with_

**Platform**
Copter

**Airframe type**
standard quadrotor X

**Hardware type**
iFlight Beast H7 55A AIO

any ideas on how to debug this?

amilcarlucas, 2022-01-03:
Why did you choose to not specify the ArduCopter firmware version?

Please update to ArduCopter 4.1.3 and retest, and make sure to use the latest mavros version as well.

And you do need to tell ArduCopter to provide the mavlink IMU messages at the rate that you need them using the SET_MESSAGE_INTERVAL mavlink message. Have you done that? Will your telemetry link have enough datarate to provide you with the mavlink messages at the rate you want them?

tiralonghipol, 2022-01-26:
@amilcarlucas 
Hi, sorry, late reply
the firmware version I just flashed via qgc is 4.1.3
what version of mavros you suggest to use to set setpoints and vision fusion?

amilcarlucas, 2022-01-27:
Like I said, the latest one.

khancyr, 2022-01-27:
Hello,

Please set the message rate  explicitly with mavros  : https://ardupilot.org/dev/docs/ros-data-from-ap.html

closed 2022-01-27 by khancyr