ArduPilot/ardupilot #19633: EKF3 uses velocity from Mavlink gps frame even if GPS_INPUT_IGNORE_FLAG_VEL_HORIZ is set.

CoolRunnerII, 2021-12-29:
## Bug report

**Issue details**

When using an external system that translates GPS and a separate positioning system to mavlink gps frames we set GPS_INPUT_IGNORE_FLAG_VEL_HORIZ since we do not calculate velocity we observe that the drone drifts off in loiter mode.

We see that the gps position is correct, but the position from EKF3 if wrong.

After setting EK3_VELNE_M_NSE and EK3_VELNE_M_NSE to 5m/s and EK3_VEL_I_GATE to 100 EKF3 started to estimate positions correctly

Expected EKF3 to disregard GPS velocity when GPS_INPUT_IGNORE_FLAG_VEL_HORIZ and not use 0.

**Version**
4.1.2

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad x

**Hardware type**
Pixhawk 4 mini

**Logs**
This is from a test flight before setting EK3_VELNE_M_NSE. 
[00000005.BIN.zip](https://github.com/ArduPilot/ardupilot/files/7790308/00000005.BIN.zip)

[image]

WickedShell, 2021-12-29:
Regarding the GPS library there is no flag in the GPS to say we don't have a velocity, we always expect a velocity from any GPS system. You can not have a vertical velocity, but the GPS library promises that if there is a GPS with a fix, then it has a velocity.

Regarding your actual error though I think you need to look at changing your `EK3_SRC1_VELXY` to not be selecting a GPS source. Because that looks to me like a request to the EKF to use the GPS as your velocity.

CoolRunnerII, 2021-12-29:
Thank you for the quick feedback!

For our case we can calculate a velocity and input it or look up your suggested parameter.

It was not obvious until after some quite scary testing that setting the ignore flag sets velocity to zero in the calculation and that this is what caused the strange behaviour.

IamPete1, 2022-09-04:
@CoolRunnerII Any update? Did changing the EKF flags resolve your issue?

peterbarker, 2024-07-31:
No answer in a couple of years, closing this one

closed 2024-07-31 by peterbarker