ArduPilot/ardupilot #19697: AP_Proximity: remove AP_Proximity_LightWareSF40C_v09

rmackay9, 2022-01-05:
We are running out of flash space on some boards so let's remove the older lightware SF40c driver. https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Proximity/AP_Proximity_LightWareSF40C_v09.h

peterbarker, 2022-01-06:
On Wed, 5 Jan 2022, Randy Mackay wrote:

> We are running out of flash space on some boards so let's remove the older lightware SF40c driver. https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Proximity/AP_Proximity_LightWareSF40C_v09.h

Shouldn't we just put it behind AP_PROXIMITY_LIGHTWARE_F40C_ENABLED - and 
default it to 0 ?

rmackay9, 2022-01-06:
@peterbarker, no, I think we should just remove it.  beyond the flash cost there is a software maintenance cost.

khanasif786, 2022-01-06:
@peterbarker @rmackay9 Please have a look at this. #19707

rmackay9, 2022-01-06:
Thanks to @khanasif786 for resolving this!

PR #19707 (2022-01-06): AP_Proximity: Removed AP_Proximity_LightWareSF40C_v09
This Fixes #19697   Deleted older drivers for fixing running out of flash space in some boards
closed 2022-01-06 by rmackay9