ArduPilot/ardupilot #19699: ArduCopter 4.2 No Mavlink Data for MinimOSD for Pixhawk 1-1M

lion13ch, 2022-01-05:
ArduCopter 4.2 "No Mavlink Data" for MinimOSD (Telem2 port used at Pixhawk 1-1M).
ArduCopter 4.1.3 works fine.

IamPete1, 2022-01-06:
@lion13ch we changed the default serial protocol to MAVLink 2, you might have to change it back to mavlink 1.

lion13ch, 2022-01-07:
At ArduCopter 4.1.3 config for SERIAL2 port serial_protocol is 2 (MAVLink 2). But it works fine. Did MAVLink 2 changed at 4.2?

IamPete1, 2022-01-07:
@lion13ch sort of, on 4.1 MAVLInk 2 was only active once a MAVLink 2 message was received. So AP was checking to make sure the other end of the connection was capable. On 4.2 we removed the check to allow streaming of MAVLink 2 on a one way link. 

So as I say, try setting it to MAVLink1.

lion13ch, 2022-01-08:
Thanks a lot!

hendjoshsr71, 2022-03-02:
@lion13ch Is your issue fixed and can we close this?

closed 2022-09-04 by IamPete1