ArduPilot/ardupilot #19702: MAV_CMD_DO_AUTOTUNE_ENABLE.param2 now includes axes to tune

hamishwillee, 2022-01-05:
## Bug report

Autotuning allows you to specify the axes to use for tuning. This is currently done using parameters. 

The latest version of [MAV_CMD_DO_AUTOTUNE_ENABLE](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_AUTOTUNE_ENABLE) allows the GCS to specify the axes as a bitmask in `param2`. Zero indicates "do all" or "do default".

The action is to update the stack to use the `MAV_CMD_DO_AUTOTUNE_ENABLE.param` values if specified and appropriate. 

This applies to all versions.

peterbarker, 2022-01-06:
One important thing to note here is that autotune in ArduPilot is seldom enabled via this command.  We have explicit modes set aside for autotuning.

Also, more a feature request that a bug report IMHO :-)

hamishwillee, 2022-01-06:
How rude :-). I've changed to a feature request to make you happy.

amilcarlucas, 2022-01-06:
Thanks for fixing the bitmask

hamishwillee, 2022-01-06:
You're welcome. Schoolboy error.

khanasif786, 2022-01-09:
@peterbarker Sir, I think this PR #19742 should work.

hendjoshsr71, 2022-03-02:
Are we going to close this issue since it was determined it wasn't worth the flash space for plane? Or leave it open for other vehicles?

rmackay9, 2022-05-12:
I'm going to close this.

closed 2022-05-12 by rmackay9