ArduPilot/ardupilot #19740: Smart Audio constant VTX messages with CRSF

stevereed66, 2022-01-09:
## Bug report

**Issue details**

Two obervations

1 ardu pilot Smart Audio is only able to change power and freq when
VTX_OPTIONS = 0 by changing VTX_BAND VTX_CHANNEL and VTX_POWER,25
Bug or  enhancement request. ?

At the bottom have included LUA script to change power on arming.

2. Get the VTX messages below as soon as Jumper TX turned on. turn off Jumper TX messages stop
eg  08/01/2022 15:28:31 : VTX: F4 5800MHz, PWR: 25mW
also VTX changing power constantly on each message between 25 and 200.

So 2 is happening on  Crossfire   6.13 Beta
roll back to Crossfire 6.10  Beta  and SA works as expected. Only vtx message on power switch.
So Not sure if this is ardupilot or crossfire problem

In the ardupilot SA wiki there is reference to
When using CRSF for both RC and VTX control it is important to disable “my VTX” support on Open TX transmitters. Follow your transmitters’ instructions.

I believe now it is VTX rx or tx set is agent X or M.
Same problem on either setting Crossfire  6.13 Beta

SYSTEM

JUMPER T16 Pro TX running EDGETX 2.5.0 

TBS CRSF  ver 6.10  Beta  and 6.13 Beta
XF MICRO TX  running CRSFSHOT
NANO RX
VTX Unify Pro HV

FC  F765-WSE

Arduplane
09/01/2022 09:57:05 : MatekF765-SE 0024003A 3056500F 20363547
09/01/2022 09:57:05 : ChibiOS: 45395b6a
09/01/2022 09:57:05 : ArduPlane V4.1.6 (eddf0367)

SmartAudio 
tx8   serial5   
SERIAL5_BAUD,4
SERIAL5_OPTIONS,4
SERIAL5_PROTOCOL,37
Wiring TX8 to white SA  Unify Pro 

AS Parms
VTX_BAND,3
VTX_CHANNEL,3
VTX_ENABLE,1
VTX_FREQ,5800
VTX_MAX_POWER,800
VTX_OPTIONS,0
VTX_POWER,25

CRSF wiring uart6 tx and rx, working as expected 
CRSF parms
brd_alt_config,1
SERIAL7_BAUD,57
SERIAL7_OPTIONS,0
SERIAL7_PROTOCOL,23

VTX Messages constant with Jumper tx on and crossfire  6.13 Beta
08/01/2022 15:28:31 : VTX: F4 5800MHz, PWR: 25mW
08/01/2022 15:28:31 : VTX: F4 5800MHz, PWR: 25mW
08/01/2022 15:28:29 : VTX: F4 5800MHz, PWR: 25mW
08/01/2022 15:28:26 : VTX: F4 5800MHz, PWR: 25mW
[... 169 lines cut ...]

end

return setVtxPower,9000

**Version**
ArduPlane V4.1.6 (eddf0367)
and 4.1.5
  
**Platform**
[  ] All
[  ] AntennaTracker
[ y ] Copter
[y  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
tri vtol agressor plane

**Hardware type**
Matek F765-WSE

**Logs**
_Please provide a link to any relevant logs that show the issue_

andyp1per, 2022-01-10:
I have a pending PR https://github.com/ArduPilot/ardupilot/pull/19497 to fix a lot of this, but requires some changes on the CRSF side

AndKe, 2022-02-27:
The message spamming is also in the dataflashlog.

stevereed66, 2022-02-27:
had this also. Backing out crsf to last non beta fixes this.I now use  the opentx agent.Andy Piper needs crsf changes before it works again