ArduPilot/ardupilot #19741: There should be parameter to give user choice when to trigger buzzer/Lost Copter Sound.

mrnams, 2022-01-09:
## Feature request

**Is your feature request related to a problem? Please describe.**
I'm always frustrated when my copter crash and no buzzer is played.
Even though I have installed "vifly finder 2", it wont get triggered when copter crashed with battery connected.

**Describe the solution you'd like**
There should be parameter to give user choice when to trigger buzzer/Lost Copter Sound.
Parameter should be like Play Lost Copter Sound. with below options
[]Battery Fail-Safe []Radio Fail-Safe []...
And user will tick particular or none or all options.

**Describe alternatives you've considered**
Right now I am trying to set radio Fail-Safe from radio to output particular channel and in Ardupilot parameter I am setting that
particular channel option as Lost Copter Sound.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Version:ArduCopter V4.1.2-rc1 (d6a4cc6f)

**Additional context**
_Add any other context or screenshots about the feature request here._

rmackay9, 2022-01-14:
There's an [auxiliary switch option](https://ardupilot.org/copter/docs/common-auxiliary-functions.html) "30" / Lost vehicle sound.  Any reason why this can't be used?

mrnams, 2022-01-20:
@rmackay9  yes it works, what i mean if it crash and no radio signal then how can we find and trigger Lost vehicle sound?
What I want is, there should be some paramter which will trigger Lost vehicle sound if it encounter any failsafe, Battery Failsafe/Radio Failsafe.

IamPete1, 2022-08-05:
@mrnams If you have any failsafe the vehicle will do lots of failsafe beeping. I'm not sure why lost vehicle beeping is better than fail safe beeping?

closed 2025-10-20 by peterbarker