ArduPilot/ardupilot #19761: after setting set_message_interval to certian rate, it reverts back after a few seconds

DoneMan, 2022-01-11:
## Bug report

**Issue details**
after connecting to sitl to mavros, i am setting "set_message_interval" to 30 hz on all topics. it increases to 30hz for about 3 seconds, but then reverts back to default.

**Version**
Copter-4.1.3

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
sitl quad

**Hardware type**
sitl board

khancyr, 2022-01-11:
hello,
Do use mavproxy (--no-mavproxy on sim_vehicle.py, or connect on tcp:5762) or disable Mavproxy streamrate control (set streamrate -1)

closed 2022-01-11 by khancyr

DoneMan, 2022-01-12:
> hello, Do use mavproxy (--no-mavproxy on sim_vehicle.py, or connect on tcp:5762) or disable Mavproxy streamrate control (set streamrate -1)

hi thanks for the help