ArduPilot/ardupilot #19803: Pixracer-periph & Neopixel NTF crash during LightsCommand

patrickpoirier51, 2022-01-15:
## Bug report

Pixracer-periph & Neopixel NTF crash during LightsCommand

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
ap_periph

**Airframe type**
n/a

**Hardware type**
PixRacer

**Logs**
_Please  tell me how to download logs AP_Periph

I have an issue with the PixRacer_Periph
I set the unit as an external NeoPixel NTF
the config is
​OUT3_FUNCTION 120
 ​OUT4_FUNCTION 121
 ​NTF_LED_TYPES 455

The Pixracer can display on a 16 Neopixel Ring but it crash and switch to maintenance after each status/color change
I can restart from GUI

 I reduced to one ring and one led , still same issue

[image]

I tried to run minimal = no sensors , still same issue

Here is parameters
ADSB_BAUDRATE,57600
ADSB_PORT,2
ARSP_DEVID,0
ARSP_PSI_RANGE,1
ARSP_TUBE_ORDER,2
ARSP_TYPE,1
BARO_ENABLE,0
BARO_EXT_BUS,-1
BARO_FLTR_RNG,0
BARO_PROBE_EXT,0
BARO1_DEVID,721682
BARO2_DEVID,0
BARO2_GND_PRESS,0
BARO3_DEVID,0
BARO3_GND_PRESS,0
BATT_MONITOR,0
BATT2_MONITOR,0
BATT3_MONITOR,0
BATT4_MONITOR,0
BATT5_MONITOR,0
BATT6_MONITOR,0
BATT7_MONITOR,0
BATT8_MONITOR,0
BATT9_MONITOR,0
BRD_SERIAL_NUM,0
CAN_BAUDRATE,1000000
CAN_NODE,99
COMPASS_AUTODEC,1
COMPASS_DEC,0
COMPASS_DEV_ID,0
COMPASS_ENABLE,0
COMPASS_EXTERNAL,0
COMPASS_FLTR_RNG,0
COMPASS_OFS_X,0
COMPASS_OFS_Y,0
COMPASS_OFS_Z,0
COMPASS_OPTIONS,0
COMPASS_ORIENT,0
COMPASS_TYPEMASK,0
COMPASS_USE,1
DEBUG,0
ESC_PWM_TYPE,0
FLASH_BOOTLOADER,0
FORMAT_VERSION,2
GPS_AUTO_CONFIG,1
GPS_COM_PORT,1
GPS_DELAY_MS,0
GPS_GNSS_MODE,0
GPS_INJECT_TO,127
GPS_MIN_DGPS,100
GPS_MIN_ELEV,-100
GPS_NAVFILTER,8
GPS_PORT,3
GPS_POS1_X,0
GPS_POS1_Y,0
GPS_POS1_Z,0
GPS_RATE_MS,200
GPS_RAW_DATA,0
GPS_SAVE_CFG,2
GPS_SBAS_MODE,2
GPS_SBP_LOGMASK,-256
GPS_TYPE,-57
NTF_BUZZ_ON_LVL,1
NTF_BUZZ_PIN,-1
NTF_BUZZ_TYPES,1
NTF_BUZZ_VOLUME,100
NTF_DISPLAY_TYPE,0
NTF_LED_BRIGHT,2
NTF_LED_LEN,1
NTF_LED_OVERRIDE,0
NTF_LED_TYPES,807
OUT_DSHOT_ESC,0
OUT_DSHOT_RATE,0
OUT_GPIO_MASK,0
OUT_RATE,50
OUT1_FUNCTION,120
OUT1_MAX,1900
OUT1_MIN,1100
OUT1_REVERSED,0
OUT1_TRIM,1500
OUT2_FUNCTION,0
OUT2_MAX,1900
OUT2_MIN,1100
OUT2_REVERSED,0
OUT2_TRIM,1500
OUT3_FUNCTION,0
OUT3_MAX,1900
OUT3_MIN,1100
OUT3_REVERSED,0
OUT3_TRIM,1500
OUT4_FUNCTION,0
OUT4_MAX,1900
OUT4_MIN,1100
OUT4_REVERSED,0
OUT4_TRIM,1500
OUT5_FUNCTION,0
OUT5_MAX,1900
OUT5_MIN,1100
OUT5_REVERSED,0
OUT5_TRIM,1500
OUT6_FUNCTION,0
OUT6_MAX,1900
OUT6_MIN,1100
OUT6_REVERSED,0
OUT6_TRIM,1500
RNGFND_BAUDRATE,115200
RNGFND_PORT,1
RNGFND1_TYPE,0

patrickpoirier51, 2022-01-16:
related
https://github.com/ArduPilot/ardupilot/pull/16113

magicrub, 2022-01-16:
Does it crash if notify is disabled via params? How about if it's not compiled in?

patrickpoirier51, 2022-01-16:
Hello Tom,

It does not crash if  its disabled = OUT1_FUNCTION to 0 or 50

_How about if it's not compiled in?_  
Do you mean that I remove define HAL_PERIPH_ENABLE_NOTIFY  and recompile ? 
What would it prove and how could we test ?

I have a question: If I enable debug, I can read on debug port (UART7 = SERIAL6)?

patrickpoirier51, 2022-01-16:
Anyway I built with  #HAL_PERIPH_ENABLE_NOTIFY and it does not crash, and obviously  no notification lights

tridge, 2022-01-26:
@patrickpoirier51 this PR should fix it:
https://github.com/ArduPilot/ardupilot/pull/19875

patrickpoirier51, 2022-01-26:
Confirmed, PR above fixed the issue
Thanks

closed 2022-01-26 by patrickpoirier51