ArduPilot/ardupilot #19827: Ac4.1 Lua scrip how to prohibit arming

roque-canales, 2022-01-18:
I m trying to prohibit arming with lua scripting

In Function of boolean variable I want to prohibit arming on multicopter.

i found : 
include AP_Arming/AP_Arming.h

singleton AP_Arming alias arming
singleton AP_Arming method disarm boolean AP_Arming::Method::SCRIPTING'literal
singleton AP_Arming method is_armed boolean
singleton AP_Arming method arm boolean AP_Arming::Method::SCRIPTING'literal
singleton AP_Arming method get_aux_auth_id boolean uint8_t'Null
singleton AP_Arming method set_aux_auth_passed void uint8_t 0 UINT8_MAX
singleton AP_Arming method set_aux_auth_failed void uint8_t 0 UINT8_MAX string

But i dont know how to use it.

does anyone can help me to achieve this ?

rmackay9, 2022-01-18:
Hi @roque-canales,

We have a couple of examples that could help

- https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/arming-check-batt-temp.lua
- https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/arming-check-wp1-takeoff.lua

I might close this issue if that's OK (but I'll still see responses and comments) because i think this is more of a developer or support question rather than an issue or enhancement request... at least for now.

closed 2022-01-18 by rmackay9

roque-canales, 2022-01-19:
Thank you perfect!

Closed #19827<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FArduPilot%2Fardupilot%2Fissues%2F19827&data=04%7C01%7C%7Ca230f657ca1b49f18e3308d9dadb98eb%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637781458959928746%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=PUhc4LE%2FkZx5EeGcdXo%2BixS%2BuPA%2B3lYw71zffYMZ%2Bgw%3D&reserved=0>.