ArduPilot/ardupilot #19839: Zero PILOT_SPEED_DN situation

standmit, 2022-01-20:
## Feature request

**Is your feature request related to a problem? Please describe.**
Special situation when PILOT_SPEED_DN = 0 isn't documented (introduced in #6951). "Oops" happens

**Describe the solution you'd like**
Special value "0" of PILOT_SPEED_DN parameter should be documented in the parameter description and maybe in AltHold  description page.

**Describe alternatives you've considered**
Alternatively, value "0" should mean "0" but not PILOT_SPEED_UP.

**Platform**
[  ] All
[  ] AntennaTracker
[ V ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
https://github.com/ArduPilot/ardupilot/blob/0af97b5bdbdc4f3e8d29e8417bea9c6e45a79317/ArduCopter/Attitude.cpp#L107-L115

standmit, 2022-05-11:
What steps need to be taken to get this updated description into the QGroundControl?

rmackay9, 2022-05-11:
@standmit,

I think 4.2 probably needs to go out as the official release.  We're almost there, perhaps 2.5 more weeks.

IamPete1, 2022-08-05:
Fix is merged and 4.2 is out.

closed 2022-08-05 by IamPete1