ArduPilot/ardupilot #19843: EKF3: support feature similar to EK2_ALT_SOURCE

ajshank, 2022-01-20:
## Feature request / Bug(?)
With EK2, the parameter `EK2_ALT_SOURCE` was useful in selecting GPS as the altitude source, and having Baro as an automatic fallback:

> `EK2_ALT_SOURCE` Primary height sensor used by the EKF. If a sensor other than Baro is selected and becomes unavailable, then the Baro sensor will be used as a fallback.

Combined with `EK2_GPS_TYPE` set to "GPS 3D vel and 2D pos", it was possible to do high-precision GPS flights with RTK setup. EKF3 in 4.1 doesn't seem to have such a straightforward setting.

Apologies if there's a combination of `EK3_` settings that achieves this. Some parts of the documentation are unclear on this, and I'm happy to edit those if required.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
My setup has two `ZED-F9P` boards, one configured as a static base; RTK fix is not a problem. However, altitude control still seems very unsteady in `POSHOLD` mode.
Setting `EK3_SRC1_POS-` parameters to GPS causes errors like "Prearm: Check `EK3_SRC3_POSZ`" (which is set to None).

rmackay9, 2022-01-20:
@ajshank,

It should be possible to do everything that was possible with EKF2 and a lot more.  So for example if you want to set the altitude source to GPS then set EK3_SRC1_POSZ = 3 (GPS).

I'm going to go ahead and close this because I think this is a support question really.  I'll monitor for replies and we can re-open if necessary.

closed 2022-01-20 by rmackay9

ajshank, 2022-01-20:
Thanks, @rmackay9. With a single IMU, one can't run more lanes (for `SRC2` or `SRC3`) -- so I'd agree this is sufficient if setting `EK3_SRC1_POSZ = 3` _also_ implies an automatic fallback to Baro. Can add to documentation if that's the case? :)

rmackay9, 2022-01-20:
@ajshank,

Yes, we can certainly improve the documentation which is here: https://ardupilot.org/copter/docs/common-ekf-sources.html

One point of confusion perhaps is the _SRC**1**_, _SRC**2**_, _SRC**3**_ selection is not related to the core.  Instead the pilot (or a lua script) can select which of the 3 source sets is used and all cores of the same source set.

ajshank, 2022-01-20:
Ah, I see -- something gave me the impression that SRC2/3 are akin to fallbacks (and thus you must have `SRC1=gps` with `SRC2=baro` for this fallback to work; but I thought SRC2 makes no sense with a single IMU).

I will consider contributing to the documentation to clarify this, especially that fallback from GPS to Baro is automatic.