ArduPilot/ardupilot #19845: There should be  an auxiliary switch option to control LED Brightness.

mrnams, 2022-01-21:
Is your feature request related to a problem? Please describe.
I'm always frustrated when my copter flash LED during day time and get recorded in my GoPro.

Describe the solution you'd like
There should be  an auxiliary switch option to control LED Brightness.

Describe alternatives you've considered
As work around I set NTF_LED_BRIGHT=0 // Off when I fly during day time,  and set NTF_LED_BRIGHT=3//High  when I fly at night.
If we control it through an auxiliary switch option, then we can control LED from radio itself instead of connecting each time to Mission Planner and modify parameter NTF_LED_BRIGHT

Platform
[ x ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

Version:ArduCopter V4.1.2-rc1 (d6a4cc6)

Additional context
Add any other context or screenshots about the feature request here.

geofrancis, 2022-02-01:
that could easily be done with a lua script.

mrnams, 2022-02-02:
> 

Can you please share link to implement that?

geofrancis, 2022-02-02:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/param_get_set_test.lua
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/RCIN_test.lua

https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_Scripting/examples

mrnams, 2022-02-03:
@geofrancis  Many thanks, I shall go through it, also can check if is there any workaround for
https://github.com/ArduPilot/ardupilot/issues/19741
Using lua script?

IamPete1, 2022-09-04:
I think this is a sufficiently unusual use-case that we can close this as possible with lua scripting.

This script should work, although I have not tested:
```
local NTF_LED_BRIGHT = Parameter()
assert(NTF_LED_BRIGHT:init('NTF_LED_BRIGHT'),'Get NTF_LED_BRIGHT failed')

local switch = assert(rc:find_channel_for_option(300),'No Scripting Switch')

function update()

  if switch:get_aux_switch_pos() == 0 then
    NTF_LED_BRIGHT:set(0)
  else
    NTF_LED_BRIGHT:set(3)
  end

  return update, 1000
end

return update()
```

closed 2022-09-04 by IamPete1