ArduPilot/ardupilot #19853: macOS:  waf Build Summary Outputs Nothing When Building Linux Boards

hendjoshsr71, 2022-01-22:
As reported on Discord by `timtuxworth` in the Infrastructure channel.

**Issue details**
When building for Linux boards on macOS users do not get any useful output from `build_summary.py`. This file [here](https://github.com/ArduPilot/ardupilot/blob/master/Tools/ardupilotwaf/build_summary.py)

"I just built plane for BeagleBone blue and got this really weird output from ./waf plane. Does anyone know what this means? BUILD SUMMARY"
```bash
Build directory: /Users/tim/ArduPilot/ardupilot/build/blue
Target         Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
--------------------------------------------------------------------------------
bin/arduplane  -         -         -        -                     - 
```

Looking at out GITHub actions  `build_summary` seems correct for CubeOrange and possibly has an issue with SITL see BSS section that seems wrong. 

CubeOrange
[image]

SITL
[image]

Building for Linux boards gives expected output on WSL2.

**Version**
Dev

**Operating System:**
macOS

timtuxworth, 2022-01-24:
Thanks Josh, I am building on new M1Pro MacBook Pro running iOS 12.1. I had to download a arm-unknown-linux-gnueabihf toolchain. I got one from here https://thinkski.github.io/osx-arm-linux-toolchains/

hendjoshsr71, 2022-01-24:
I'm not sure the best place to get the compilers for linux for macOs.....

I was going to tell you to please install the the recommended compiler listed [here ](https://firmware.ardupilot.org/Tools/STM32-tools/ )instead. But then remembered you are using a Linux board ;)

timtuxworth, 2022-02-05:
I just built master (4 Feb 2022) and now I'm not getting a proper build summary for any of my boards. It looks like it might be building for SITL regardless what I configure it for. (I did waf distclean then waf configure board=X) each time:

PixRacer (why does it say SITL? I definitely built for pixracer)
BUILD SUMMARY
Build directory: /Users/tim/ArduPilot/ardupilot/build/sitl
Target         Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
--------------------------------------------------------------------------------
bin/arduplane   2392064    180224        0               2572288  Not Applicable

PixHawk1-1M
BUILD SUMMARY
Build directory: /Users/tim/ArduPilot/ardupilot/build/sitl
Target         Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
--------------------------------------------------------------------------------
bin/arduplane   2392064    180224        0               2572288  Not Applicable

Durandal
BUILD SUMMARY
Build directory: /Users/tim/ArduPilot/ardupilot/build/sitl
Target         Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
--------------------------------------------------------------------------------
bin/arduplane   2392064    180224        0               2572288  Not Applicable