ArduPilot/ardupilot #19856: When GPS Lost, the fail safe don't work fine!

vmv007, 2022-01-23:
## Bug report

**when the quad loses gps signal in auto mode, it will switch to althold mode and start flying fast, and do not brakes**
**Version**
all stable firmwares but 4.2.0 dev  working good
**Platform**

[  ] Copter

**Airframe type**
Quad

**Hardware type**
Pixhawk

IamPete1, 2022-01-23:
We need a log to help diagnose this. https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html#downloading-logs-via-mavlink

vmv007, 2022-01-23:
> We need a log to help diagnose this. https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html#downloading-logs-via-mavlink

Ok, but i already deleted log((( i will go to fly and get a new one

vmv007, 2022-01-23:
> > We need a log to help diagnose this. https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html#downloading-logs-via-mavlink
> 
> https://drive.google.com/file/d/1jpons_JSS_THixdD0-Bnsg_jJUgLydYp/view?usp=drivesdk

IamPete1, 2022-01-23:
That log is from 4.0.7, the current stable copter release is 4.1.3, please see if the issue remains on 4.1.3.

I'm not really sure what the issue is your having, looks like it switches to alt hold due to the EKF failsafe and then descends because the pilot input throttle is low? What behavior are you expecting?

vmv007, 2022-01-23:
> That log is from 4.0.7, the current stable copter release is 4.1.3, please see if the issue remains on 4.1.3.
>
> I'm not really sure what the issue is your having, looks like it switches to alt hold [...]

other firmware same shit!
when althold is enabled, it should slow down and not accelerate!

IamPete1, 2022-01-23:
In alt hold desired roll and pitch are zero. 

[image]

I see no evidence of it accelerating in alt hold.

vmv007, 2022-01-23:
> In alt hold desired roll and pitch are zero. 
> 
> [image]
> 
> I see no evidence of it accelerating in alt hold. 

the copter was flying on a slope and did not slow down when it switched to althold, i will try one more and will sent log here! When will be good weather

IamPete1, 2022-01-23:
OK, I think I understand. The problem is not that is was trying to accelerate due to some failsafe action, just that it lost track of which way is up. We do see a difference in pitch between EKF and dcm.

[image]

Hard to tell which was right of course....

I suspect the difference you saw between 4.2 and 4.0 is EKF3 vs EKF2. 

Maybe you can test 4.1 with EKF3. 

I don't think there is a huge amount we can do in the code (not easily anyway). You could try and reduce your vibrations, that should help.

rmackay9, 2022-01-24:
Rather than vibration I think it might be a GPS issue. If we look at the number of satellites and accuracies (see GPA message) it looks like it completely loses lock.
[image]

By the way, in AltHold mode the pilot controls the lean angle so it won't slow down on it's own.

At this point I think this issue is really a support issue.  Sorry to be strict but this should really be raised in the support forums first and then when we've confirmed it's a real bug or enhancement request we raise an issue here.  If we don't then our to-do list turns into a support forum.

I'll still be monitoring this thread for replies though.  Txs for the report.

closed 2022-01-24 by rmackay9

vmv007, 2022-01-24:
> Rather than vibration I think it might be a GPS issue. If we look at the number of satellites and accuracies (see GPA message) it looks like it completely loses lock.
> [image]
>
> By the way, in AltHold [...]

I used jamming gps, there are no problems with gps

vmv007, 2022-02-04:
[image]