ArduPilot/ardupilot #19871: AP_BattMonitor: sum monitor should handle generators and solar panel controllers

rmackay9, 2022-01-25:
It would be nice if the "Sum" battery monitor handled generators and solar panel controllers by subtracting the their current from the total rather than adding it.

One way to accomplish this would be to add yet another parameter, this time a "BATTx_SUB_MASK" or "BATTx_INV_MASK" bitmask that allows specifying which battery monitor's current should be subtracted.

FYI @magicrub

magicrub, 2022-01-25:
Love it! The param BATTx_INV_MASK would live in the _SUM param domain

hendjoshsr71, 2022-01-25:
Do the generators and MPPTs not return negative current in case of generating energy? If not, that would be the fix I think would be best. From my perusal it seems like they should be able to but I haven't traced every path.

I know for the analog monitors you can setup them up in sitl to be "chargers" and they seem to work as you'd expect. IE you gain capacity. 

Of course doing your suggestion would be easier for systems where someone has "flipped" the physical to sensor such that charging (negative) currents show as positive on GCSs.

rmackay9, 2022-01-25:
@hendjoshsr71,

The MPPT I have from PrecisionDigital reports positive currents.  @magicrub and I discussed this and for my specific case I could use the new CURR_MULT parameter to switch it to be negative but this issue was a more general purpose change that would work even if an analog battery monitor was used with a solar panel charge controller.

WickedShell, 2022-01-25:
> more general purpose change that would work even if any type of battery monitor was used with a solar panel charge controller

I think the point is this actually *does* just work for analog monitors. Simply set the scale to be negative instead of positive, and you will see it add to the battery capacity over time.

rmackay9, 2022-01-26:
ok, @WickedShell sounds correct to me.  So instead of a new INV_MASK parameter we need to change the AP_BattMonitor_Generator and MPPT (part of BattMonjitor_UAVCAN) so they report negative currents.

@peterbarker how do you feel about this?  (I think you wrote much of the generator code).

hendjoshsr71, 2022-01-26:
> If MPPT is the wrong direction then we should be flipping the sign bit appropriately.

This is what I was trying to get across. In power systems like ours negative current generally means charging. You can actually "charge" your solar panels..... (they are diodes after all) and can waste energy if your system setup is wrong.

The best fix would be to change the devices, I'm guessing the number of these out there are quite small ...... but realize that is unlikely. 

Rant below...
 `dsdl/mppt/200020.Stream.uavcan.` [here ](https://github.com/dronecan/DSDL/blob/master/mppt/20020.Stream.uavcan) is an example where the documentation says nothing about meaning of positive vs negative currents. Thus a different manufacturer could interpret it differently.