ArduPilot/ardupilot #19887: Sensor driver issues on Cube Orange

Murcamil, 2022-01-26:
**Issue details**
Hi,

I have upgraded from AC 4.0.5 to 4.1.3 and for some reason I cannot find, no sensor is being read by the H7 CPU, as far as I understand. I tried: 

- Upload ArduRover/ArduPlane and then back ArduCopter
- Update bootloader (successful) and then do the above
- Checked the Cube internally and the IMU board is connected to the main board
- Checked all flight logs of the board and it did not have any crash
- Uploaded latest PX4 and all looks fine. Loaded ArduCopter afterwards and got the same results.

All ArduPilot versions appear to have the same issue. Before the upgrade, everything was fine.
[There were others with a similar if not identical issue.](https://discuss.ardupilot.org/t/cubeblack-can-not-detect-its-baro/79045/11)

Thank you

**Version**
ArduCopter V4.1.3 (2fb939a1)

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Not related. Tarot 580

**Hardware type**
CubeOrange

**Logs**
[Here is a TLog.](https://drive.google.com/file/d/1TYYByFw65RSXp2hqEKh56gjXqpcfaj7y/view?usp=sharing)

peterbarker, 2022-01-27:
@proficnc for visibility

tridge, 2022-01-27:
@Murcamil I can't access the tlog, can you please give access to tridge60@gmail.com

proficnc, 2022-01-27:
lets continue this on the forum at the link you provided, If we find a software bug, we can come back here...

Murcamil, 2022-01-27:
The issue was hardware related.

closed 2022-01-27 by Murcamil