ArduPilot/ardupilot #19894: TECS Controller: Integrator throttle state

berickso, 2022-01-27:
## Bug report

**Issue details**

_Please describe the problem_
TECS Controller "_integTHR_state" is not calculated appropriately in the loop as written.  I have confirmed this discrepancy over a TCP/UDP connection to Simulink and plotting the variables from SITL simulation in real time. 3 methods were used and plotted with respect to the integrator limits:
   1) the raw integrator throttle state output from the source code "AP_integTHR_state" 
   2) the product of the 4 input variables plus the previous time step 1) "int_THR_state_sum"
   3) the loop recreated in Simulink from the addition of the previous time step out of the loop "int_THR_state_loop"

Detailed examination of the data on over a short time step reveals the recreated loop 3) shows the output scales appropriately with the change of magnitude of the product of the four input values at each time step.

The source code output 1) cannot be mathematically recreated as it is written.

2) is equivalent to 1) but is of questionable accuracy as 1) cannot be recreated.

**Version**
4.1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
[image]
[image]
[image]
[image]
[image]
[image]
[image]

tridge, 2022-02-01:
@priseborough need your input on this one

tridge, 2022-02-01:
@berickso can you provide the logs you based this on?

berickso, 2022-02-01:
> @berickso can you provide the logs you based this on?
Hello, thanks for looking into this!

I did not record logs in Mission Planner for this particular simulation but, the problem can be replicated 100% of the time and I am happy to run additional simulations, provide logs, and any additional info that is helpful.  I have not created a log before so it may take some time tomorrow to do so. I can log the simulink data as well and provide that in a useful file format (is a .mat ok?).

For your reference, this is a SITL simulation just booting mission planner and using all default parameters.

Thanks,
Brett

berickso, 2022-02-01:
[AP_SITL_simulation.zip](https://github.com/ArduPilot/ardupilot/files/7980635/AP_SITL_simulation.zip)
[image]

Here is a zip with logs from a comparable simulation and the logs created by mission planner.  I've logged the data in Simulink as a couple different matlab file types.  Hope these are of use.  Let me know if there is anything else I can provide.

berickso, 2022-02-01:
[image]
[image]
[image]
[image]

For background info, I am developing a Simulink-based, Arduplane control system blockset for use in a flight simulator that leverages mission planner to issue navigation controls for the model and implements 6DoF physics via CFD characterization of the aircraft.  I can validate every segment of the model (with some room for small improvement) except the integrator throttle state variable.  In that particular block, all inputs have been validated against the source code.  The issue is isolated to the aforementioned loop which appears to be both mathematically correct and equivalent with how the variable is defined in the source code.

berickso, 2022-02-17:
Following up to see if there is any new information or insight on this issue.

Thanks,
Brett

Naterater, 2022-02-17:
This makes sense as to why I **consistently** see large altitude/pitch swings at the START of a landing sequence.