ArduPilot/ardupilot #19938: GCS_Common: `set_message_interval` missing response target

ES-Alexander, 2022-01-30:
## Bug report

[`MAV_CMD_SET_MESSAGE_INTERVAL`](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL) is supposed to support a `response target` in parameter 7, but the [ArduPilot implementation doesn't handle that parameter](https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L2555).

This isn't a feature I need to use, just something that was found while discussing some external documentation.

**Version**
All

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2022-01-31:
On Sun, 30 Jan 2022, ES-Alexander wrote:

> MAV_CMD_SET_MESSAGE_INTERVAL is supposed to support a response target in parameter 7, but the ArduPilot implementation doesn't handle that parameter.
> 
> This isn't a feature I need to use, just something that was found while discussing some external documentation.

Can you think of a message it would be useful with?

What happens if two other systems request a message on the same link? 
Send it to both at different rates over the same link?

Yeah, not useful.  PX4 don't appear to use it either, on a cursory 
inspection.  They do appear to have ascribed meaning to param3 without 
telling anyone about it, OTOH....

olliw42, 2022-01-31:
this parameter IMHO has quite some significant importance. You should not think only about simple mavlink systems, like a radio link and a ardupilot, like it seems many of you constantly do, but a mavlink system with some components in it. A usual example: A mavlink gimbal, or a mavlink camera.

In such configurations you DO want an option to establish a communication between two components of a system which is ONLY between these two components. Eg., the gimbal might want to get a message stream from the autopilot at a relatively high rate in order to do its thing well, and also vice versa the autopilot might want to get a message stream from the gimbal at a relatively high rate. However, these messages should not be communicated across the whole system or even worse across all systems. 

Concluding it is not useful is shortsighted. Not surprisingly, with ArduPilot, whenever one wants to get serious with mavlink, one often encounters that the whole mavlink network is spilled with irrelevant data, obviously putting stress on the whole system, including the radio link, for no good reason.

This is a very useful parameter, and it would be a great relieve if you would consider supporting it.

IMHO.

peterbarker, 2022-02-02:
On Mon, 31 Jan 2022, olliw42 wrote:

> This is a very useful parameter, and it would be a great relieve if you would consider supporting it.

Can you give an example of a streaming message where the target would be 
useful?