ArduPilot/ardupilot #19963: RTC initialization using uninitialized value

shellixyz, 2022-02-01:
https://github.com/ArduPilot/ardupilot/pull/19824/commits/f21db73a29a182ec93089a57b42b7402f2d91d2e in master introduced a bug. The RTC is now initialized using an uninitialized value.

`AP_GPS::update_instance()` calls `AP_GPS::time_epoch_usec()` to set the RTC. The problem is that `AP_GPS::time_epoch_usec()` uses `drivers[instance]->get_last_itow()` which returns `_last_itow` before it has been assigned.

I don't know exactly how to fix that

shellixyz, 2022-02-01:
```
        fix_time_ms = time_epoch_convert(istate.time_week, istate.time_week_ms) * 1000ULL;
        return (fix_time_ms + (AP_HAL::millis() - istate.last_gps_time_ms)) * 1000ULL;
```

That's also wrong, should be

```
        fix_time_ms = time_epoch_convert(istate.time_week, istate.time_week_ms);
        return (fix_time_ms + (AP_HAL::millis() - istate.last_gps_time_ms)) * 1000ULL;
```

bugobliterator, 2022-02-02:
> [f21db73](https://github.com/ArduPilot/ardupilot/commit/f21db73a29a182ec93089a57b42b7402f2d91d2e) from #19824 in master introduced a bug. The RTC is now initialized using an uninitialized value.
> 
> `AP_GPS::update_instance()` calls `AP_GPS::time_epoch_usec()` to set the RTC. The problem is that `AP_GPS::time_epoch_usec()` uses `drivers[instance]->get_last_itow()` which returns `_last_itow` before it has been assigned.
> 
> I don't know exactly how to fix that. @bugobliterator ?

Hi @shellixyz you are right about the mistake of multiplying time_epoch_convert with 1000ULL, but on the second point, I don't believe that should happen. in the method `AP_GPS::time_epoch_usec` we check `istate.last_corrected_gps_time_us != 0` last_corrected_gps_time_us is set in the method `AP_GPS_Backend::check_new_itow` alongside _last_itow. So its always going to else condition when _last_itow is not initialised.

Can you please share details on how you are able to reproduce the issue, so I can look into it further. Thanks for reporting this.

bugobliterator, 2022-02-02:
On the first issue, it will be visible on UAVCAN and MAV GPS types. So needs fixing there.

shellixyz, 2022-02-02:
To reproduce I just ran SITL and watched the timestamp of the last log which is randomly from 1 month to several months ahead of the actual date/time then traced the issue back to `drivers[instance]->get_last_itow()`. I have first noticed the issue on a wing running master on which the timestamp of the logs didn't match the data/time of the flights

shellixyz, 2022-02-02:
We are February 2nd but see the timestamp of the last log:

[image]

bugobliterator, 2022-02-02:
@shellixyz I have pushed the fix in the PR, it should be working now.

closed 2022-02-02 by tridge