ArduPilot/ardupilot #19982: Integrator not reset once landed in autonomous landing modes

bnsgeyer, 2022-02-03:
## Bug report
When landing in any of the autonomous landing modes (i.e. RTL, Land) the integrator will freeze on land_complete_maybe however it will never reset until spool state reaches ground idle.  This is now a problem for heli's with the new integrator scheme that uses land complete to reset the integrator.  If a heli were to land with this new integrator scheme in an autonomous mode, it would maintain the frozen integrator from when it reached land_complete_maybe until the aircraft disarmed.  This will cause undue stress on the rotor system.

The bug is in the make_safe_ground_handling (4.1 on) or make_safe_spool_down (4.0).  The resetting of the integrator and yaw axis should apply in all spool states as whenever this method is used the aircraft is disarmed or landed.

**Version**
4.0
4.1
master

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Heli

**Hardware type**
Found in SITL

[image]
In this plot, it can be seen that the pitch and roll integrator is never reset

lthall, 2022-02-03:
Is there any reason we can't run these lines here:
https://github.com/ArduPilot/ardupilot/blob/08438c30a6cea636d48a0c431e5f0346b9fff011/ArduCopter/mode.cpp#L501

`        attitude_control->reset_rate_controller_I_terms_smoothly();
        attitude_control->reset_yaw_target_and_rate();`

In all spool states. Or maybe we reset the I term in all states but leave the yaw?

bnsgeyer, 2022-02-03:
Yes we leave yaw and reset the integrator. Here is a branch I am getting ready to PR
https://github.com/bnsgeyer/ardupilot/tree/pr-auto-lndg-int-fix

hendjoshsr71, 2022-03-02:
I think this was addressed with the linked PR? If I'm wrong reopen please.

closed 2022-03-02 by hendjoshsr71