ArduPilot/ardupilot #19996: SITL: Can't load default parameters from an absolute path

WickedShell, 2022-02-04:
## Bug report

**Issue details**

Loading a set of default parameters from an absolute path will fail, but a relative path to the parameters will succeed.

``` bash
[wickedshell@lackey ardupilot]$ /home/wickedshell/code/ardupilot/build/sitl/bin/arduplane -S --model quadplane --speedup 1 --slave 0 --defaults /home/wickedshell/code/ardupilot/Tools/autotest/default_params/quadplane.parm -I0
Setting SIM_SPEEDUP=1.000000
Suggested EK3_BCOEF_* = 16.288, EK3_MCOEF = 0.209
Starting sketch 'ArduPlane'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
Connection on serial port 5760
PANIC: Failed to load defaults from /home/wickedshell/code/ardupilot/Tools/autotest/default_params/quadplane.parm

Running: sh Tools/scripts/dumpstack.sh 80186 >dumpstack.sh_arduplane.80186.out 2>&1
dumpstack.sh has been run.  Output was:
-------------- begin dumpstack.sh output ----------------
Tools/scripts/dumpstack.sh: line 14: gdb: command not found
-------------- end dumpstack.sh output ----------------
Running: sh Tools/scripts/dumpcore.sh 80186 >dumpcore.sh_arduplane.80186.out 2>&1
dumpcore.sh has been run.  Output was:
-------------- begin dumpcore.sh output ----------------
Tools/scripts/dumpcore.sh: line 16: gdb: command not found
-------------- end dumpcore.sh output ----------------
^C
[wickedshell@lackey ardupilot]$ ls -l /home/wickedshell/code/ardupilot/Tools/autotest/default_params/quadplane.parm
-rw-r--r-- 1 wickedshell wickedshell 1854 Feb  4 14:25 /home/wickedshell/code/ardupilot/Tools/autotest/default_params/quadplane.parm
[wickedshell@lackey ardupilot]$ /home/wickedshell/code/ardupilot/build/sitl/bin/arduplane -S --model quadplane --speedup 1 --slave 0 --defaults Tools/autotest/default_params/quadplane.parm -I0
Setting SIM_SPEEDUP=1.000000
Suggested EK3_BCOEF_* = 16.288, EK3_MCOEF = 0.209
Starting sketch 'ArduPlane'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
Connection on serial port 5760
Loaded defaults from Tools/autotest/default_params/quadplane.parm
Home: -35.363262 149.165237 alt=584.000000m hdg=353.000000
Smoothing reset at 0.001
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
validate_structures:469: Validating structures
Loaded defaults from Tools/autotest/default_params/quadplane.parm
```

**Version**
bb1bd69172d708

**Platform**
[ x ] All (Tested against plane, probably all though)
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

hendjoshsr71, 2022-02-04:
@Williangalvani

Williangalvani, 2022-02-05:
somehow this was being bypassed before my patch:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Filesystem/AP_Filesystem_posix.cpp#L39

Can you check if removing this helps?

I'm not sure what was the reasoning behind it, though...

shiv-tyagi, 2022-02-05:
Have been getting same error [since](https://github.com/ArduPilot/ardupilot/commit/e6929892e36405b62cc6a002b85b4a09ba27356d) this commit was merged. 
[image]

WickedShell, 2022-10-14:
This got fixed somewhere along the way, I can't replicate it on master ( afe7790e4b9f )

closed 2022-10-14 by WickedShell