ArduPilot/ardupilot #20025: Why did my drone crash? EKF3 IMU0 emergency yaw reset + GPS Glitch

framosnt, 2022-02-07:
## Bug report

After configure and did some perfectly flights, he get "EKF3 IMU0 emergency yaw reset" after "GPS Glitch" and invert direction turned upside down and fell! 

after it broke for the first time, I fixed everything and changed the controller and kept only the GPS module, I believed it was accelerometer failures. I did the flight test and it crashed the same way again!!!!

**Version**
Copter 4.1.1 - customized.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad big drone

**Hardware type**
CUAV X7+ and GPS NEO V2

**Logs**
https://drive.google.com/drive/folders/1bL7zrH1y6gtbgjIm4fPRHyjaFKqLEnjd?usp=sharing

amilcarlucas, 2022-02-07:
Why are you using 4.1.1? Why do you not rebase your customizations on top of 4.1.3?

framosnt, 2022-02-08:
@rmackay9 thank you !